ostaquet/Arduino-SIM800L-driver #2: Can't compile example sketches

Bartvelp, 2019-07-19:
Hi!
Using the arduino compiler I am not able to compile the example sketches. I have attached below the error message. The library looks pretty well written, certainly a lot better than my own mess, so I would love to use it.
```
Arduino:1.8.9 (Windows 10), Board:"NodeMCU 1.0 (ESP-12E Module), 160 MHz, Flash, Disabled, Basic SSL ciphers (lower ROM use), 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 256000"

C:\Users\user\Documents\Arduino\libraries\Arduino-SIM800L-driver-master\SIM800L.cpp: In constructor 'SIM800L::SIM800L(int, int, int, unsigned int, unsigned int, bool)':

C:\Users\user\Documents\Arduino\libraries\Arduino-SIM800L-driver-master\SIM800L.cpp:35:45: error: invalid conversion from 'void*' to 'char*' [-fpermissive]

   internalBuffer = malloc(internalBufferSize);

                                             ^

C:\Users\user\Documents\Arduino\libraries\Arduino-SIM800L-driver-master\SIM800L.cpp:43:37: error: invalid conversion from 'void*' to 'char*' [-fpermissive]

   recvBuffer = malloc(recvBufferSize);

                                     ^

C:\Users\user\Documents\Arduino\libraries\Arduino-SIM800L-driver-master\SIM800L.cpp: In member function 'int SIM800L::doPost(const char*, const char*, char*, unsigned int, unsigned int)':

C:\Users\user\Documents\Arduino\libraries\Arduino-SIM800L-driver-master\SIM800L.cpp:65:32: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]

   int initRC = initiateHTTP(url);

                                ^

In file included from C:\Users\user\Documents\Arduino\libraries\Arduino-SIM800L-driver-master\SIM800L.cpp:8:0:

C:\Users\user\Documents\Arduino\libraries\Arduino-SIM800L-driver-master\SIM800L.h:69:9: error:   initializing argument 1 of 'int SIM800L::initiateHTTP(char*)' [-fpermissive]

     int initiateHTTP(char* url);

         ^

C:\Users\user\Documents\Arduino\libraries\Arduino-SIM800L-driver-master\SIM800L.cpp:71:54: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]

   sendCommand("AT+HTTPPARA=\"CONTENT\",", contentType);

                                                      ^

In file included from C:\Users\user\Documents\Arduino\libraries\Arduino-SIM800L-driver-master\SIM800L.cpp:8:0:

C:\Users\user\Documents\Arduino\libraries\Arduino-SIM800L-driver-master\SIM800L.h:54:10: error:   initializing argument 2 of 'void SIM800L::sendCommand(char*, char*)' [-fpermissive]

     void sendCommand(char* command, char* parameter);

          ^

C:\Users\user\Documents\Arduino\libraries\Arduino-SIM800L-driver-master\SIM800L.cpp:78:27: error: invalid conversion from 'void*' to 'char*' [-fpermissive]

   char* tmpBuf = malloc(30);

                           ^

C:\Users\user\Documents\Arduino\libraries\Arduino-SIM800L-driver-master\SIM800L.cpp:158:27: error: ISO C++ forbids comparison between pointer and integer [-fpermissive]

         if((recvBuffer == '\r') || (recvBuffer == '\n')) {

                           ^

C:\Users\user\Documents\Arduino\libraries\Arduino-SIM800L-driver-master\SIM800L.cpp:158:51: error: ISO C++ forbids comparison between pointer and integer [-fpermissive]

         if((recvBuffer == '\r') || (recvBuffer == '\n')) {

                                                   ^

C:\Users\user\Documents\Arduino\libraries\Arduino-SIM800L-driver-master\SIM800L.cpp: In member function 'int SIM800L::doGet(const char*, unsigned int)':

C:\Users\user\Documents\Arduino\libraries\Arduino-SIM800L-driver-master\SIM800L.cpp:203:32: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]

   int initRC = initiateHTTP(url);

                                ^

In file included from C:\Users\user\Documents\Arduino\libraries\Arduino-SIM800L-driver-master\SIM800L.cpp:8:0:

C:\Users\user\Documents\Arduino\libraries\Arduino-SIM800L-driver-master\SIM800L.h:69:9: error:   initializing argument 1 of 'int SIM800L::initiateHTTP(char*)' [-fpermissive]

     int initiateHTTP(char* url);

         ^

C:\Users\user\Documents\Arduino\libraries\Arduino-SIM800L-driver-master\SIM800L.cpp:268:27: error: ISO C++ forbids comparison between pointer and integer [-fpermissive]

         if((recvBuffer == '\r') || (recvBuffer == '\n')) {

                           ^

C:\Users\user\Documents\Arduino\libraries\Arduino-SIM800L-driver-master\SIM800L.cpp:268:51: error: ISO C++ forbids comparison between pointer and integer [-fpermissive]

         if((recvBuffer == '\r') || (recvBuffer == '\n')) {

                                                   ^

Meerdere bibliotheken gevonden voor "SoftwareSerial.h"
Gebruikt: C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial
Niet gebruikt: C:\Users\user\Documents\Arduino\libraries\EspSoftwareSerial
exit status 1
Fout bij het compileren voor board NodeMCU 1.0 (ESP-12E Module)

Dit rapport zou meer informatie bevatten met
"Uitgebreide uitvoer weergeven tijden compilatie"
optie aan in Bestand -> Voorkeuren.

```

ostaquet, 2019-08-02:
Hi,
Sorry for the delay, I just coming back from annual leave ;-)
Your issue is confusing for me. It seems that standard C++ commands are not recognised.
To be sure, I updated my Arduino IDE to 1.8.9 and compile is working well for official Arduino boards.

I'm not familiar with ESP8266 boards and maybe somebody else could help.

KR,
Olivier

Bartvelp, 2019-08-02:
Hi!
I can confirm the library compiles for standard arduino boards as well for me. But I must say it suprises me. I tried to look in to the errors generated and selected this one:
```
C:\Users\user\Documents\Arduino\libraries\Arduino-SIM800L-driver-master\SIM800L.cpp:35:45: error: invalid conversion from 'void*' to 'char*' [-fpermissive]

   internalBuffer = malloc(internalBufferSize);
```

It seems to me that this is just invalid C++ code. According to [this](http://www.cplusplus.com/reference/cstdlib/malloc/) and [this](https://stackoverflow.com/questions/18879344/c-error-invalid-conversion-from-void-to-char). The return pointer should be explicty cast to char* like so:
`internalBuffer = (char*) malloc(internalBufferSize);`
This change also seems to remove the error generated when compiling for ESP8266. I am by no means a C++ programmer though and could be wrong. (I also have no idea how to fix other error generated)
The strange thing for is that this compiles for another board and doesn't throw errors, maybe it is related to [this](https://github.com/arduino/Arduino/wiki/Build-Process#compilation) part of the arduino docs.

ostaquet, 2019-08-06:
Maybe you can adjust the parameters of the compiler for the ESP8266?
In the stack trace, compiler suggests to use the "-fpermissive" option...

ostaquet, 2019-08-25:
Hi @Bartvelp 
I just tried to fix the compile problems based on your input.
Can you checkout the last version and try it?
Can you give me the compile report?

ostaquet, 2019-08-25:
I just compiled the last version with the NodeMCU 1.0 (ESP-12E Module) and it worked.
The code is still running well on the Arduino Mega2560.
It should be OK now 😉 
Can you confirm @Bartvelp ?

Bartvelp, 2019-08-25:
Now getting:
```

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
Test sketch

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
Test sketch
```
Hi I just tried your latest update. It compiles fine now! just getting a watchdog reset. This is a thing related to the wifi capabilities of the ESP. If you do not call `yield()` or `delay()` for 3 (?) seconds the ESP will reset.
This is from the HTTPS_GET test sketch, adding delay or yield in a loop should be an easy fix.

ostaquet, 2019-08-26:
Fixed in the v1.1.0

closed 2019-08-26 by ostaquet