ostaquet/Arduino-SIM800L-driver #3: Board: "WAVGAT UNO R3"

grfbr, 2019-08-24:
Hi ostaquet!

First of all! thanks to sharing your code with the community! 

Well, unfortunately, I don't know code in C++ so i can't help too much hahaha, but basically, I'm trying to compile the code to a clone of Arduino, however, when it tries to compile and upload the code, occurs some errors (below). Could you try to help me to figure out what's happening? It's my IDE? The Board? The libraries? 

Thanks a lot!

Arduino: 1.8.9 (Mac OS X), Board: "WAVGAT UNO R3"

```
Warning: platform.txt from core 'Arduino AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
/Users/mac/Documents/Arduino/libraries/Arduino-SIM800L-driver-master/SIM800L.cpp: In constructor 'SIM800L::SIM800L(int, int, int, unsigned int, unsigned int, bool)':
/Users/mac/Documents/Arduino/libraries/Arduino-SIM800L-driver-master/SIM800L.cpp:35:26: error: invalid conversion from 'void*' to 'char*' [-fpermissive]
   internalBuffer = malloc(internalBufferSize);
                          ^
/Users/mac/Documents/Arduino/libraries/Arduino-SIM800L-driver-master/SIM800L.cpp:43:22: error: invalid conversion from 'void*' to 'char*' [-fpermissive]
   recvBuffer = malloc(recvBufferSize);
                      ^
/Users/mac/Documents/Arduino/libraries/Arduino-SIM800L-driver-master/SIM800L.cpp: In member function 'int SIM800L::doPost(const char*, const char*, char*, unsigned int, unsigned int)':
/Users/mac/Documents/Arduino/libraries/Arduino-SIM800L-driver-master/SIM800L.cpp:65:32: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
   int initRC = initiateHTTP(url);
                                ^
In file included from /Users/mac/Documents/Arduino/libraries/Arduino-SIM800L-driver-master/SIM800L.cpp:8:0:
/Users/mac/Documents/Arduino/libraries/Arduino-SIM800L-driver-master/SIM800L.h:69:9: note:   initializing argument 1 of 'int SIM800L::initiateHTTP(char*)'
     int initiateHTTP(char* url);
         ^
/Users/mac/Documents/Arduino/libraries/Arduino-SIM800L-driver-master/SIM800L.cpp:71:54: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
   sendCommand("AT+HTTPPARA=\"CONTENT\",", contentType);
                                                      ^
In file included from /Users/mac/Documents/Arduino/libraries/Arduino-SIM800L-driver-master/SIM800L.cpp:8:0:
/Users/mac/Documents/Arduino/libraries/Arduino-SIM800L-driver-master/SIM800L.h:54:10: note:   initializing argument 2 of 'void SIM800L::sendCommand(char*, char*)'
     void sendCommand(char* command, char* parameter);
          ^
/Users/mac/Documents/Arduino/libraries/Arduino-SIM800L-driver-master/SIM800L.cpp:78:24: error: invalid conversion from 'void*' to 'char*' [-fpermissive]
   char* tmpBuf = malloc(30);
                        ^
/Users/mac/Documents/Arduino/libraries/Arduino-SIM800L-driver-master/SIM800L.cpp:158:27: error: ISO C++ forbids comparison between pointer and integer [-fpermissive]
         if((recvBuffer == '\r') || (recvBuffer == '\n')) {
                           ^
/Users/mac/Documents/Arduino/libraries/Arduino-SIM800L-driver-master/SIM800L.cpp:158:51: error: ISO C++ forbids comparison between pointer and integer [-fpermissive]
         if((recvBuffer == '\r') || (recvBuffer == '\n')) {
                                                   ^
/Users/mac/Documents/Arduino/libraries/Arduino-SIM800L-driver-master/SIM800L.cpp: In member function 'int SIM800L::doGet(const char*, unsigned int)':
/Users/mac/Documents/Arduino/libraries/Arduino-SIM800L-driver-master/SIM800L.cpp:203:32: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
   int initRC = initiateHTTP(url);
                                ^
In file included from /Users/mac/Documents/Arduino/libraries/Arduino-SIM800L-driver-master/SIM800L.cpp:8:0:
/Users/mac/Documents/Arduino/libraries/Arduino-SIM800L-driver-master/SIM800L.h:69:9: note:   initializing argument 1 of 'int SIM800L::initiateHTTP(char*)'
     int initiateHTTP(char* url);
         ^
/Users/mac/Documents/Arduino/libraries/Arduino-SIM800L-driver-master/SIM800L.cpp:268:27: error: ISO C++ forbids comparison between pointer and integer [-fpermissive]
         if((recvBuffer == '\r') || (recvBuffer == '\n')) {
                           ^
/Users/mac/Documents/Arduino/libraries/Arduino-SIM800L-driver-master/SIM800L.cpp:268:51: error: ISO C++ forbids comparison between pointer and integer [-fpermissive]
         if((recvBuffer == '\r') || (recvBuffer == '\n')) {
                                                   ^
exit status 1
Error compiling for board WAVGAT UNO R3.
```

ostaquet, 2019-08-25:
Hum...
It seems that you encountered the same issue than @Bartvelp (issue #2).
I tried to reproduce but without having an ESP or a Wavgat boards, it is not easy...

Bartvelp, 2019-08-25:
Maybe you can add the ESP8266 boards tot your board manager in Arduino. Then when you select any board (like nodemcu) you can just hit compile to reproduce. @ostaquet

ostaquet, 2019-08-25:
Hi guys,
I just tried to fix the compile problems based on your input (completely blind 🙈  as it compile for the boards I own)
Can you checkout the last version and try it?
Can you give me the compile report?

ostaquet, 2019-08-25:
I just compiled the last version with the NodeMCU 1.0 (ESP-12E Module) and it worked.
The code is still running well on the Arduino Mega2560.
It should be OK now 😉 
Can you confirm?

ostaquet, 2019-08-26:
Fixed in v1.1.0

closed 2019-08-26 by ostaquet

grfbr, 2019-08-27:
Hi ostaquet! First of all! thanks for your help! 

I will try to compile this today and let you know as soon as possible! 

I'm having a problem with my SIM800L and I believe that the issue is my power supply 🙄

grfbr, 2019-08-27:
Hi Olive! Yes! Is compiling with no problem! Amazing! 

Now the issue is my SIM800L hahaha, but i will figure out what's is going on!

Again, thank you very much mate!

Just a suggestion: Could you try to work with MQTT protocol?

Cheers!