ostaquet/Arduino-SIM800L-driver #4: Watchdog triggered during usage on ESP8266

ostaquet, 2019-08-26:
Now getting:
```

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
Test sketch

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
Test sketch
```
Hi I just tried your latest update. It compiles fine now! just getting a watchdog reset. This is a thing related to the wifi capabilities of the ESP. If you do not call `yield()` or `delay()` for 3 (?) seconds the ESP will reset.
This is from the HTTPS_GET test sketch, adding delay or yield in a loop should be an easy fix.

_Originally posted by @Bartvelp in https://github.com/ostaquet/Arduino-SIM800L-driver/issues/2#issuecomment-524642956_

ostaquet, 2019-08-26:
Hi @Bartvelp,
The `delay()` functions are used quite often in the driver to manage the delays for establish GDPR communication and so on.
I made some experiment with the `avr/wdt.h` library (watchdog management for AVR) but it doesn't recognise the `yield()`command or consider the usage of `delay()`as yielding.
After reading the code again, there is no active waiting. The `delay()` calls are used properly.
Did you try to only remove the `while(!Serial);` in the beginning of the examples?
It should be enough.
KR,
Olivier

g-Romano, 2021-02-22:
hi,
i'm using version 1.13.2 on wemos d1 mini and i'm having the same problem, there is no solution at the moment?

udayshankarkanuri, 2023-01-28:
> Hi @Bartvelp, The `delay()` functions are used quite often in the driver to manage the delays for establish GPRS communication and so on. I made some experiment with the `avr/wdt.h` library (watchdog management for AVR) but it doesn't recognise the `yield()`command or consider the usage of `delay()`as yielding. After reading the code again, there is no active waiting. The `delay()` calls are used properly. Did you try to only remove the `while(!Serial);` in the beginning of the examples? It should be enough. KR, Olivier
i use esp 32 and sim 800l to sends the HTTPS POST request to clould function.i remove while(!Serial):but no use.its automatically resets and shows this type of error.

14:35:36.486 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
14:35:36.486 -> Memory dump at 0x400d1028: ffffffff ffffffff ffffffff
14:35:36.486 -> Core 1 register dump:
14:35:36.486 -> PC      : 0x400d102c  PS      : 0x00060530  A0      : 0x800d0ec9  A1      : 0x3ffb1f60  
14:35:36.486 -> A2      : 0x3ffb8668  A3      : 0x3ffbfe5c  A4      : 0x00000006  A5      : 0x000000c8  
14:35:36.486 -> A6      : 0x00000200  A7      : 0x00000000  A8      : 0x800d10b8  A9      : 0x3ffb1f40  
14:35:36.512 -> A10     : 0x3ffb8668  A11     : 0x00000000  A12     : 0x3ffbdc7c  A13     : 0x00000010  
14:35:36.512 -> A14     : 0x00000011  A15     : 0x00000008  SAR     : 0x0000001a  EXCCAUSE: 0x00000000  
14:35:36.512 -> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
14:35:36.512 -> 
14:35:36.512 -> ELF file SHA256: 0000000000000000
14:35:36.545 -> 
14:35:36.545 -> Backtrace: 0x400d102c:0x3ffb1f60 0x400d0ec6:0x3ffb1f80 0x400d2c96:0x3ffb1fb0 0x40086155:0x3ffb1fd0

please help me to find out the solution

ostaquet, 2024-06-02:
As mentioned in the README, the project is no longer maintained as my module is broken. Thanks for reporting the issue.

closed 2024-06-02 by ostaquet