ostaquet/Arduino-SIM800L-driver #5: MQTT protocol implementation

ostaquet, 2019-08-27:
Hi Olive! Yes! Is compiling with no problem! Amazing! 

Now the issue is my SIM800L hahaha, but i will figure out what's is going on!

Again, thank you very much mate!

Just a suggestion: Could you try to work with MQTT protocol?

Cheers!

_Originally posted by @grfbr in https://github.com/ostaquet/Arduino-SIM800L-driver/issues/3#issuecomment-525107454_

ostaquet, 2019-08-27:
Hi,
At the moment, there is no plan to implement MQTT protocol in the driver. I took good note of your suggestion ;-)
KR

ankitmhatre, 2020-12-18:
Hey @ostaquet, I think it is already a high time for having few MQTT functions.

ostaquet, 2021-01-17:
Hi @ankitmhatre, you are free to contribute ;-). There are some ideas to implement it in issue #30.

ostaquet, 2024-06-02:
As mentioned in the README, the project is no longer maintained as my module is broken. Thanks for reporting the issue.

closed 2024-06-02 by ostaquet