ostaquet/Arduino-SIM800L-driver #7: More functionality?

PanzerLlama, 2020-01-13:
Happy to find a maintained SIM800 lib. I just tried it and would like to ask if it would be possible to add some extra functionality, not sure if that is in the scope of this lib but I need a location information (GPRS not GPS) so I added the following function:

void SIM800L::getLocation(char* lat, char* lng) {
  sendCommand_P(AT_CMD_CLBS);
  if (readResponse(DEFAULT_TIMEOUT)) {
  
    int16_t idxLatStart = strIndex(internalBuffer, "+CLBS: ");
    int16_t idxLatEnd = strIndex(internalBuffer, ",", idxLatStart + 9);
    int16_t idxLngStart = idxLatEnd + 1;
    int16_t idxLngEnd = strIndex(internalBuffer, ",", idxLngStart);
    
    int index = 0;
	for(int16_t i = (idxLatStart + 9); i < (idxLatEnd); i++) {
		lat[index++] = internalBuffer[i];
	}
	lat[index] = '\0';
	
	index = 0;
	for(int16_t i = (idxLngStart); i < (idxLngEnd); i++) {
		lng[index++] = internalBuffer[i];
	}
	lng[index] = '\0';
  }
}

My c++ skills are quite poor so I believe it is not a candidate for a pull request, however maybe you could consider adding the location to status functions?

ostaquet, 2020-01-20:
Thanks for sharing this.
I will review the code and see how to integrate it.

The idea of the library is to manage HTTP/HTTPS connectivity and to keep it as lean and simple as possible. The SIM800L module is offering a lot of features and I try not to have a big library to save memory on small micro controllers.

PanzerLlama, 2020-01-20:
Sure thing, that's what I suspected (i.e. out of scope) and I totally get your point. If you'd consider an "add-on" with some extra functionality that would be great though. I went through several sim800 libs and most are not maintained and\or have serious issues (being too big is one of them). I personally use your lib to send some telemetry through http and getting the approximate location of the device is very useful.

ostaquet, 2020-05-23:
I was not able to retrieve coordinates from my network provider. Without the ability to test the process, I prefer to let it out of the library.

Seems that it is moving feature from SIMCom because on my firmware (R14), the command is `AT+CIPGSMLOC` and required the connection to the bearer. At the end, it gave me only the timestamp but no coordinates (probably a GDPR/Privacy stuff...)

closed 2020-05-23 by ostaquet