ostaquet/Arduino-SIM800L-driver #9: Newbie: 'setupModule' was not declared in this scope

tcdanh, 2020-04-28:
I'm newbie from Vietnam. I would like to setup an irrigation project with Arduino Uno and module SIM800A, the users will control via website (internet). First I am test module with Uno through the AT command -> ok. 
Then I downloaded the SIM800L library, and testing with the HTTPS_GET_SoftSerial example (code). I have 02 issues:
- sim800l = new SIM800L((Stream *)serial, SIM800_RST_PIN, 200, 512); --> I don't know where SIM800_RST_PIN on my Module SIM800A?
- After copying the code, I try to compile but it announce:" Arduino: 1.8.10 (Windows 7), Board: "Arduino Uno"
C:\Users\Vaio\Documents\Innovation_club\Iot\Irrigration_System\Arduino\sketch_oct15a\GMS_V1\GMS_V1.ino: In function 'void setup()':
GMS_V1:30:3: error: 'setupModule' was not declared in this scope
   setupModule();
Multiple libraries were found for "SoftwareSerial.h"
 Used: C:\Users\Vaio\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\libraries\SoftwareSerial
Multiple libraries were found for "SIM800L.h"
 Used: C:\Users\Vaio\Documents\Arduino\libraries\Arduino-SIM800L-driver-1.8.0
exit status 1
'setupModule' was not declared in this scope
This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Could you help us to solve this? Thank you so much

ostaquet, 2020-05-01:
Did you try to use the examples as-in?

chiggs002, 2020-05-01:
Hi Olivier, I also tried your library with a SIM800L chip. 

Specifically the chips is:  SIM800L V2 EVB chip.
[](https://www.ebay.com/p/23023423047)

I am trying to send data via Software Serial to a server with https requests, and it returned a 702 error. See details below.

Also does the SIM800L chip support SSL? When I talk to the chip with "AT" commands and issue the command : AT+HTTPSSL=? it returns an "ERROR". It does not exist? 

Do "AT+HTTPSSL" commands exist on the SIM800L V2 chip?

I used the code from your existing library examples.  " https_get_softserial".

The reponse I get:

Setup Complete!
Signal OK (strenght: 12)
Network registration OK
GPRS config OK
GPRS connected !
Start HTTP GET...
HTTP GET error 702
GPRS disconnected !
Module in low power mode

Any help appreciated Olivier. Thank you

See the screen shot below:
[image]
 error.

tcdanh, 2020-05-04:
Thank you, now it works because that's my fault: missing function

closed 2020-05-04 by ostaquet