ostaquet/Arduino-SIM800L-driver #10: Usage of library with a SIM800L chip

ostaquet, 2020-05-01:
Hi Olivier, I also tried your library with a SIM800L chip. 

Specifically the chips is:  SIM800L V2 EVB chip.https://www.ebay.com/p/23023423047
[](https://www.ebay.com/p/23023423047)

I am trying to send data via Software Serial to a server with https requests, and it returned a 702 error. See details below.

Also does the SIM800L chip support SSL? When I talk to the chip with "AT" commands and issue the command : AT+HTTPSSL=? it returns an "ERROR". It does not exist? 

Do "AT+HTTPSSL" commands exist on the SIM800L V2 chip?

I used the code from your existing library examples.  " https_get_softserial".

The reponse I get:

Setup Complete!
Signal OK (strenght: 12)
Network registration OK
GPRS config OK
GPRS connected !
Start HTTP GET...
HTTP GET error 702
GPRS disconnected !
Module in low power mode

Any help appreciated Olivier. Thank you

See the screen shot below:
[image]
 error.

_Originally posted by @chiggs002 in https://github.com/ostaquet/Arduino-SIM800L-driver/issues/9#issuecomment-622351738_

ostaquet, 2020-05-01:
@chiggs002 can you enable the full debug and post the result?

To do it, just update the constructor in the example:
`sim800l = new SIM800L((Stream *)serial, SIM800_RST_PIN, 200, 512);`
 Has to be modified to:
`sim800l = new SIM800L((Stream *)serial, SIM800_RST_PIN, 200, 512, (Stream *)&Serial);`

chiggs002, 2020-05-02:
Hi Olivier , I modified the code as you asked:

My results:

v⸮⸮&R⸮⸮⸮⸮⸮⸮B⸮SIM800L : Reset
SIM800L : Prepare internal buffer of 200 bytes
SIM800L : Prepare reception buffer of 512 bytes
SIM800L : Send "AT"
SIM800L : Receive timeout
Problem to initialize AT command, retry in 1 sec
SIM800L : Send "AT"
SIM800L : End of transmission
SIM800L : Receive "AT

OK
"
Setup Complete!
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ

+CSQ: 0,0
"
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ

+CSQ: 13,0
"
Signal OK (strenght: 13)
SIM800L : Send "AT+CREG?"
SIM800L : End of transmission
SIM800L : Receive "AT+CREG?

+CREG: 0,2
"
SIM800L : Send "AT+CREG?"
SIM800L : End of transmission
SIM800L : Receive "AT+CREG?

+CREG: 0,1
"
Network registration OK
SIM800L : Send "AT+SAPBR=3,1,"Contype","GPRS""
SIM800L : End of transmission
SIM800L : Receive "AT+SAPBR=3,1,"Contype","GPRS"

OK
"
SIM800L : Send "AT+SAPBR=3,1,"APN","data.uk""
SIM800L : End of transmission
SIM800L : Receive "AT+SAPBR=3,1,"APN","data.uk"

OK
"
GPRS config OK
SIM800L : Send "AT+SAPBR=1,1"
SIM800L : End of transmission
SIM800L : Receive "AT+SAPBR=1,1

OK
"
GPRS connected !
Start HTTP GET...
SIM800L : Send "AT+HTTPINIT"
SIM800L : End of transmission
SIM800L : Receive "AT+HTTPINIT

OK
"
SIM800L : Send "AT+HTTPPARA="CID",1"
SIM800L : End of transmission
SIM800L : Receive "AT+HTTPPARA="CID",1

OK
"
SIM800L : Send "AT+HTTPPARA="URL","www.google.com""
SIM800L : End of transmission
SIM800L : Receive "AT+HTTPPARA="URL","www.google.com"

OK
"
SIM800L : Send "AT+HTTPSSL=0"
SIM800L : End of transmission
SIM800L : Receive "AT+HTTPSSL=0

ERROR
"
SIM800L : initiateHTTP() - Unable to switch to HTTP
HTTP GET error 702
SIM800L : Send "AT+SAPBR=0,1"
SIM800L : End of transmission
SIM800L : Receive "AT+SAPBR=0,1

OK
"
GPRS disconnected !
SIM800L : Send "AT+CFUN?"
SIM800L : End of transmission
SIM800L : Receive "AT+CFUN?

+CFUN: 1
"
SIM800L : Send "AT+CFUN=0"
SIM800L : Send "AT+CFUN?"
SIM800L : End of transmission
SIM800L : Receive "AT+CFUN=0
AT+CFUN?

+CFUN: 0
"

chiggs002, 2020-05-02:
Olivier, is the error of the chip that it requires a firmware upgrade?

ostaquet, 2020-05-03:
Hi Chiggs,
I just uploaded BasicTest in the examples folder.
Can you launch this sketch and send me the output?

chiggs002, 2020-05-03:
Yes of course

ostaquet, 2020-05-03:
On my side, I received:
```
Start of test protocol
Problem to initialize AT command, retry in 1 sec
Module ready
Module SIM800 R14.18
Firmware Revision:1418B05SIM800L24
Signal OK (strenght: 14)
Network registration OK
End of test protocol
```

chiggs002, 2020-05-03:
Just ran it:

⸮⸮⸮⸮⸮Start of test protocol
Problem to initialize AT command, retry in 1 sec
Module ready
Module SIM800 R13.08
Firmware Revision:1308B08SIM800L16
Signal OK (strenght: 13)
Network registration OK
End of test protocol

chiggs002, 2020-05-03:
Olivier. What chip are u using can you point me to source ? Please Thank you

ostaquet, 2020-05-04:
Damned. The version of your module is probably the source of the problem.

I just find this on the TinyGSM library https://github.com/vshymanskyy/TinyGSM/issues/29#issuecomment-328802556.

> On my SIM800 R14.18 it works fine.
> The other one, SIM800 R13.08 reports that SSL is not supported.

I will add some test on the library to be sure that the version issue is spotted quickly ( #11 ).

Regarding the sourcing, I ordered mine on Amazon: https://www.amazon.fr/dp/B073TF2QKL (the big advantage is that the serial TTL supports 5V ;-) )

closed 2020-05-04 by ostaquet

stegeron, 2022-04-08:
Hi Olivier, look, I have the same version 14.18 as you do and still I get the HTTP error 702... Do you have any other clues where to search the bug?

20:46:43.660 -> Start of test protocol
20:46:48.666 -> Problem to initialize AT command, retry in 1 sec
20:46:49.743 -> Module ready
20:46:49.743 -> Module SIM800 R14.18
20:46:49.743 -> Firmware Revision:1418B04SIM800L24
20:46:49.790 -> SIM status READY
20:46:49.837 -> SIM card number 8942001450338567978f
20:46:51.997 -> Signal OK (strenght: 18)
20:46:54.057 -> Network registration OK
20:46:54.057 -> End of test protocol

Thank you, Ondrej

ostaquet, 2022-05-01:
Can you make a test about which encryption protocol are supported by your web server?

You can test it with Qualys SSL Labs : https://www.ssllabs.com

stegeron, 2022-05-02:
Hello, you can see the whole report here: https://testtls.com/steger.cz/443

Thanks, O.

ostaquet, 2022-05-08:
According to the test result, the server doesn't offer SSL2, SSL3 or TLS 1.0...

As you can see in the [Security concerns section](https://github.com/ostaquet/Arduino-SIM800L-driver#security-concerns), the SIM800L supports only SSL2, SSL3 or TLS 1.0.

DmitriyVovk, 2022-10-17:
`\xFFProblem to initialize AT command, retry in 1 sec
AT
OK
Setup Complete!
ATI
SIM800 R14.18

OK
Module SIM800 R14.18
Firmware ReviAT+GMR
Revision:1418B04SIM800L24

OK
sion:1418B04SIM800L24
SIM status READY
SAT+CPIN?
+CPIN: READY

OK
AT+CCID
8938003992894053808f

OK
IM card number 8938003992894053808f
AT+CSQ
+CSQ: 0,0

OK

Call Ready

SMS Ready
AT+CSQ
+CSQ: 0,0

OK
AT+CSQ
+CSQ: 24,0

OK
Signal OK (strenght: 24)
AT+CREG?
+CREG: 0,1

OK
Network registration OK
GPRAT+SAPBR=3,1,"Contype","GPRS"
OK
AT+SAPBR=3,1,"APN","internet"
OK
S config OK
GPRS connected with IP 
OK
AT+SAPBR=2,1
+SAPBR: 1,1,"10.99.139.103"

OK
10.99.139.103
Start HTTP GET...
AT+HTTPINIT
OK
AT+HTTPPARA="CID",1
OK
AT+HTTPPARA="URL","http://mcu-bin-upd.000webhostapp.com/bin/example.bin"
OK
HTTP GET error 702
AT+SAPBR=0,1
OK
GPRS disconnected !
Module AT+CFUN?
+CFUN: 1

OK
AT+CFUN?0
+CFUN: 0

OK
in low power mode

+CPIN: NOT READY

OK`

Hello dear codders. Please help. I have the same error. But one strange thing: i had success GET request to the file. May be i add some minor! changes after. Sometime i have spikes on power that cause restarting module and today i was fighting with it.
Now it says 702... 
The same server, no changes was happen on it. The file is available from browser. Thanks for reading..