ostaquet/Arduino-SIM800L-driver #11: Add testing of version to run the HTTPS

ostaquet, 2020-05-04:
According to some tests and issues  (#10 and https://github.com/vshymanskyy/TinyGSM/issues/29), the version 13.08 doesn't support SSL connectivity.

In order to avoid troubles in the future, add the check before trying to establish a communication.

andrewstillie, 2020-05-05:
I am using a TTGO-Tcall module.  I am using the Arduino ecosystem to write my code with 
TinyGsm
TinyGsmClient
HttpClient

How can I request the SIM800 version number?

ostaquet, 2020-05-05:
You can install the last version of the library and run the BasicTest example ;-)

andrewstillie, 2020-05-05:
Thank You - I can see I have SIM800 R14.18
How do I activate SSL for my POST to AWS API Gateway?

ostaquet, 2020-05-23:
Did you already try the examples provided with the library? (like [HTTPS_POST_HardwareSerial](https://github.com/ostaquet/Arduino-SIM800L-driver/blob/master/examples/HTTPS_POST_HardwareSerial/HTTPS_POST_HardwareSerial.ino))

You may encounter some security issues as the SIM800L is not so recent and doesn't support the latest SSL security layers (like some 605 or 606 errors as documented in the [SSL application notes](https://github.com/ostaquet/Arduino-SIM800L-driver/blob/master/extras/SIM800%20Series_SSL_Application%20Note_V1.00.pdf))

ostaquet, 2020-05-23:
Issue fixed with release 1.11.0

closed 2020-05-23 by ostaquet