ostaquet/Arduino-SIM800L-driver #12: http request with login and password

mamanbudiman, 2020-05-17:
It is great to have this library. It is worked.
I have a server cloud that use login and password for http request.

Do you know how to implement the login and password ?

Regards,

ostaquet, 2020-05-23:
If you're using the HTTP Basic Authentication, you have to add the HTTP header in the request (as described on [Wikipedia Basic access authentication](https://en.wikipedia.org/wiki/Basic_access_authentication)).

You can also try to add the parameters in the URL like `http://username:password@www.example.com` (not tested).

ostaquet, 2020-05-23:
The `http://username:password@www.example.com` doesn't work.

You should add this in the user data through the AT command `AT+HTTPPARA` as below (example with `user` and `password` as credentials :

`AT+HTTPPARA=“USERDATA”,“Authorization:Basic dXNlcjpwYXNzd29yZA==”`

Where `dXNlcjpwYXNzd29yZA==` is the BASE64 encoding of `user:password`

closed 2020-05-23 by ostaquet