ostaquet/Arduino-SIM800L-driver #14: Problem to initialize AT command, retry in 1 sec

fabioce, 2020-07-07:
Hello
I'm trying your code HTTPS_POST_SoftSerial with an ESP32 with SIM800L embed (TTGO T-Call)
I set correctly my PINS and My APN
#define SIM800_TX_PIN 27
#define SIM800_RX_PIN 26
#define SIM800_RST_PIN 5
But I have always this error:
Problem to initialize AT command, retry in 1 sec
Have you suggestions for me?
Thanks

ostaquet, 2020-07-07:
I quickly searched on Google and it seems that the ESP32 has 3 hardware serial available. I suggest to use serial 1 or serial 2 ;-). I also know that the Arduino Mega has some limitations about which pins can be used for SoftwareSerial. It is maybe the same with the ESP32 but I don't have any experiences with the ESP32.

If you have the error mentioned, it is because the serial didn't connect to the SIM800L. Check exchanging the pin 27 and 26 (sometimes, we messed up the TX and RX :-P) and check the wiring.

fabioce, 2020-07-07:
Hi
now I changed with HardwareSerial and only these settings differ to yours:

#define SIM800_TX_PIN 27
#define SIM800_RX_PIN 26
#define SIM800_RST_PIN 5

Serial1.begin(9600, SERIAL_8N1, SIM800_RX_PIN, SIM800_TX_PIN);

But the result is the same.
I cannot change the PINS because the SIM800L is embed:
https://www.amazon.it/TTGO-T-Call-Wireless-Antenna-SIM800L/dp/B07VFPBRJ4/

ostaquet, 2020-07-07:
The hardware serial doesn't need to assign pins.

Please test with the [BasicTest_HardwareSerial.ino](https://github.com/ostaquet/Arduino-SIM800L-driver/blob/master/examples/BasicTest_HardwareSerial/BasicTest_HardwareSerial.ino) sketch on Serial1.

I saw in the repository of @Xinyuan-LilyGO that the Arduino code is running on the hardware serial 1 (see https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/blob/master/examples/Arduino_TinyGSM/Arduino_TinyGSM.ino).

fabioce, 2020-07-09:
Hello,
I copied your basictest and deployed into my device, here the result:

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d0dd0: ffffffff ffffffff ffffffff
Core 1 register dump:
PC      : 0x400d0dd4  PS      : 0x00060130  A0      : 0x800d0c7d  A1      : 0x3ffb1f60  
A2      : 0x3ffb864c  A3      : 0x3ffc02c8  A4      : 0x00000006  A5      : 0x000000c8  
A6      : 0x00000200  A7      : 0x00000000  A8      : 0x800d0e60  A9      : 0x3ffb1f40  
A10     : 0x3ffb864c  A11     : 0x00000000  A12     : 0x3ffbebfc  A13     : 0x00000009  
A14     : 0x0000000a  A15     : 0x00000004  SAR     : 0x0000001a  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d0dd4:0x3ffb1f60 0x400d0c7a:0x3ffb1f80 0x400d1a3f:0x3ffb1fb0 0x40088215:0x3ffb1fd0

fabioce, 2020-07-09:
Now I changed this PIN
#define SIM800_RST_PIN 5
and now I have
Problem to initialize AT command, retry in 1 sec

ostaquet, 2020-07-15:
After a quick search on Google, I found a tutorial on how to publish data with the module you have ([see here](https://randomnerdtutorials.com/esp32-sim800l-publish-data-to-cloud/)).

The interesting thing is that it seems that the power management is manage on the chip by the ESP32 on specific pins.

If you looks the code from the tutorial, those parts are interesting:
```
// TTGO T-Call pins
#define MODEM_RST            5
#define MODEM_PWKEY          4
#define MODEM_POWER_ON       23
#define MODEM_TX             27
#define MODEM_RX             26
// Set serial for AT commands (to SIM800 module)
#define SerialAT Serial1
```

It seems that the reset pin is on the pin 5 and there are additional power management to handle on pins 4 and 23 to power on the SIM800L. It also confirm that the SIM800L is connected through the hardware serial (`Serial1`).

As seen later in the code (`setup()`):
```
  // Set modem reset, enable, power pins
  pinMode(MODEM_PWKEY, OUTPUT);
[...]
  pinMode(MODEM_POWER_ON, OUTPUT);
  digitalWrite(MODEM_PWKEY, LOW);
[...]
  digitalWrite(MODEM_POWER_ON, HIGH);
```

So, my guess (because I don't have the module available) is that you should modify the `BasicTest_HardwareSerial.ino` as below:

```
#include "SIM800L.h"

#define SIM800_RST_PIN     5
#define SIM800_PWKEY     4
#define SIM800_POWER_ON     23

SIM800L* sim800l;

void setup() {  
  // Initialize Serial Monitor for debugging
  Serial.begin(115200);
  while(!Serial);

  // Set modem reset, enable, power pins
  pinMode(SIM800_PWKEY, OUTPUT);
  pinMode(SIM800_POWER_ON, OUTPUT);
  digitalWrite(SIM800_PWKEY, LOW);
  digitalWrite(SIM800_POWER_ON, HIGH);

  // Initialize the hardware Serial1
  Serial1.begin(9600);
  delay(1000);
   
  // Initialize SIM800L driver with an internal buffer of 200 bytes and a reception buffer of 512 bytes, debug disabled
  // sim800l = new SIM800L((Stream *)&Serial1, SIM800_RST_PIN, 200, 512);

  // Equivalent line with the debug enabled on the Serial
  sim800l = new SIM800L((Stream *)&Serial1, SIM800_RST_PIN, 200, 512, (Stream *)&Serial);

  Serial.println("Start of test protocol");

  // Wait until the module is ready to accept AT commands
  while(!sim800l->isReady()) {
    Serial.println(F("Problem to initialize AT command, retry in 1 sec"));
    delay(1000);
  }

  Serial.println("Module ready");

  // Print version
  Serial.print("Module ");
  Serial.println(sim800l->getVersion());
  Serial.print("Firmware ");
  Serial.println(sim800l->getFirmware());

  // Wait for the GSM signal
  uint8_t signal = sim800l->getSignal();
  while(signal <= 0) {
    delay(1000);
    signal = sim800l->getSignal();
  }

  if(signal > 5) {
    Serial.print(F("Signal OK (strenght: "));
  } else {
    Serial.print(F("Signal low (strenght: "));
  }
  Serial.print(signal);
  Serial.println(F(")"));
  delay(1000);

  // Wait for operator network registration (national or roaming network)
  NetworkRegistration network = sim800l->getRegistrationStatus();
  while(network != REGISTERED_HOME && network != REGISTERED_ROAMING) {
    delay(1000);
    network = sim800l->getRegistrationStatus();
  }
  Serial.println(F("Network registration OK"));

  Serial.println("End of test protocol");
}
 
void loop() {
  
}
```

EmanueleFeola, 2020-07-24:
Hi! I had the same problem with TTGO T-Call and I solved it by modifying the function getSignal in SIM800L.cpp (1) and by adding the pin configuration at the beginning of the .ino file (2)
(I can copy and paste the whole files if needed)

(1)
The problem was that the string into internalBuffer containing the signal strength after readResponse is "+CSQ: 0,0", BUT before the '+' character there also are CR and LF characters, hence i changed the if condition to idxBase != 2 instead of idxBase != 0

uint8_t SIM800L::getSignal() {
  sendCommand_P(AT_CMD_CSQ);
  if(readResponse(DEFAULT_TIMEOUT)) {
    //int16_t idxBase = strIndex(internalBuffer, "AT+CSQ");
    int16_t idxBase = strIndex(internalBuffer, "+CSQ");
    if(idxBase != 2) {
		Serial.printf("idxBase %d\n", idxBase);
      return 0;
    }
    int16_t idxEnd = strIndex(internalBuffer, ",", idxBase);
    uint8_t value = internalBuffer[idxEnd - 1] - '0';
    if(internalBuffer[idxEnd - 2] != ' ') {
      value += (internalBuffer[idxEnd - 2] - '0') * 10;
    }
    if(value > 31) {
      return 0;
    }
    return value;
  }
  return 0;
}

(2)

// TTGO T-Call pins
#define MODEM_RST            5
#define MODEM_PWKEY          4
#define MODEM_POWER_ON       23
#define MODEM_TX             27
#define MODEM_RX             26
#define I2C_SDA              21
#define I2C_SCL              22
.
.
.

// Set modem reset, enable, power pins
pinMode(MODEM_PWKEY, OUTPUT);
pinMode(MODEM_RST, OUTPUT);
pinMode(MODEM_POWER_ON, OUTPUT);
digitalWrite(MODEM_PWKEY, LOW);
digitalWrite(MODEM_RST, HIGH);
digitalWrite(MODEM_POWER_ON, HIGH);

// Initialize the hardware Serial1
SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);
delay(1000);

closed 2020-09-25 by ostaquet