ostaquet/Arduino-SIM800L-driver #15: WolfSSL for SIM800L

SciWax, 2020-07-17:
Dear ostaquet,

first thanks for your library.  I have a problem with the security in regards of a modern backend of a server, which you've mentioned here: https://github.com/ostaquet/Arduino-SIM800L-driver#security-concerns

I get the error `606`, when I want to send my data to a webserver. I've potentially found information about how to get the WolfSSL library on the Arduino. I guess I also have to put a new certificate on the SIM800L? There is some information about how to do it potentially, but which certificate do I have to use? One which is provided by WolfSSL I guess?

Did you personally already try to make a SIM800L work with WolfSSL, when  `606` or `605` pops up?

ostaquet, 2020-07-19:
There are two main issues with the IP stack of the SIM800L.

First, the certificates embedded on the SIM800L are quite outdated (January 2016) and should be updated. By uploading new certificates, you can solve this issue... but not the next one...

Second, the encryption technologies embedded on the SIM800L are completely outdated (SSL2, SSL3 and TLS1.0) and many cloud infrastructures (AWS, GCP...) don't accept those old protocol anymore for security reasons.

I didn't use personally WolfSSL. WolfSSL is a library to support the SSL layer and especially the most recent implementation like TLS1.2 or TLS1.3. I guess that WolfSSL has a mechanism to keep the certificates up-to-date (should be normal).

I'm afraid that if you want to make the SIM800L working with WolfSSL, you will need to use the lower part of the IP stack of the SIM800L (HTTP with WolfSSL on top or even at TCP level...). The support team of WolfSSL will probably be able to support you as it is a commercial product.

closed 2020-07-19 by ostaquet

SciWax, 2020-07-19:
Thank you for answering the question. Guess I need an alternative to that modem with an up-to-date IP-Stack....