ostaquet/Arduino-SIM800L-driver #16: HTTPS_GET_SoftSerial.ino gets stuck after successfull Get request [200]

SciWax, 2020-07-23:
Dear ostaquet,

when I try out your example with a custom URL for writing data into my database, I get an issue of the code being stuck everytime I successfully realize a get request (Code 200).

Let me show you my changes compared to your example code:

```
#define SIM800_TX_PIN 12
#define SIM800_RX_PIN 13
#define SIM800_RST_PIN 2

const char APN[] = "internet.telekom";
const char URL[] = "https://url.com/a.php?t1=1.22&AdaBat=4.1&ModemBat=3.8&h1=4.4&t2=5.4&h2=9.4&o31=2.0&no21=3.0&co1=9.0&o32=10&no22=5&co2=10&PM1=9.4&PM2=1.1&PM4=14.12&PM10=14.12";

// Equivalent line with the debug enabled on the Serial
  sim800l = new SIM800L((Stream *)serial, SIM800_RST_PIN, 300, 512, (Stream *)&Serial);
```
The real custom url is 36 characters long.

The output in debugging mode:

```
15:41:59.846 -> SIM800L : Reset
15:42:01.843 -> SIM800L : Prepare internal buffer of 256 bytes
15:42:01.843 -> SIM800L : Prepare reception buffer of 512 bytes
15:42:01.843 -> SIM800L : Send "AT"
15:42:06.836 -> SIM800L : Receive timeout
15:42:06.836 -> Problem to initialize AT command, retry in 1 sec
15:42:07.822 -> SIM800L : Send "AT"
15:42:07.857 -> SIM800L : End of transmission
15:42:07.857 -> SIM800L : Receive "AT

15:42:07.857 -> OK
15:42:07.857 -> "
15:42:07.857 -> Setup Complete!
15:42:07.857 -> SIM800L : Send "AT+CSQ"
15:42:07.889 -> SIM800L : End of transmission
15:42:07.889 -> SIM800L : Receive "AT+CSQ

15:42:07.889 -> +CSQ: 0,0
15:42:07.889 -> "
15:42:08.909 -> SIM800L : Send "AT+CSQ"
15:42:08.943 -> SIM800L : End of transmission
15:42:08.943 -> SIM800L : Receive "AT+CSQ

15:42:08.943 -> +CSQ: 0,0
15:42:08.943 -> "
15:42:09.964 -> SIM800L : Send "AT+CSQ"
15:42:09.998 -> SIM800L : End of transmission
15:42:09.998 -> SIM800L : Receive "AT+CSQ

15:42:09.998 -> +CSQ: 17,0
15:42:09.998 -> "
15:42:09.998 -> Signal OK (strenght: 17)
15:42:10.984 -> SIM800L : Send "AT+CREG?"
15:42:11.016 -> SIM800L : End of transmission
15:42:11.050 -> SIM800L : Receive "AT+CREG?

15:42:11.050 -> +CREG: 0,1
15:42:11.050 -> "
15:42:11.050 -> Network registration OK
15:42:12.038 -> SIM800L : Send "AT+SAPBR=3,1,"Contype","GPRS""
15:42:12.107 -> SIM800L : End of transmission
15:42:12.107 -> SIM800L : Receive "AT+SAPBR=3,1,"Contype","GPRS"

15:42:12.139 -> OK
15:42:12.139 -> "
15:42:12.139 -> SIM800L : Send "AT+SAPBR=3,1,"APN","internet.telekom""
15:42:12.208 -> SIM800L : End of transmission
15:42:12.208 -> SIM800L : Receive "AT+SAPBR=3,1,"APN","internet.telekom"
[... 198 lines cut ...]
16:07:16.999 -> "
16:07:16.999 -> GPRS disconnected !
16:07:16.999 -> SIM800L : Send "AT+CFUN?"
16:07:17.033 -> SIM800L : End of transmission
16:07:17.033 -> SIM800L : Receive "AT+CFUN?

16:07:17.067 -> +CFUN: 1
16:07:17.067 -> "
16:07:17.067 -> SIM800L : Send "AT+CFUN=0"
16:07:17.067 -> SIM800L : Send "AT+CFUN?"
16:07:17.101 -> SIM800L : End of transmission
16:07:17.101 -> SIM800L : Receive "AT+CFUN=0
AT+CFUN?

16:07:17.135 -> +CFUN: 0
16:07:17.135 -> "
16:07:17.135 -> Module in low power mode

```

I've looked through your `SIM800L.cpp` file, but couldnt really find a solution. It feels like that I'm stuck in a while loop (maybe), but I'm not sure.

Kind regards

SciWax

SciWax, 2020-07-23:
Commenting this part in the `.cpp` didn't work for me.

```
 /*
    // Read number of bytes defined in the dataSize
    for(uint16_t i = 0; i < dataSize && i < recvBufferSize; i++) {
      while(!stream->available());
      if(stream->available()) {
        // Load the next char
        recvBuffer[i] = stream->read();
        // If the character is CR or LF, ignore it (it's probably part of the module communication schema)
        if((recvBuffer[i] == '\r') || (recvBuffer[i] == '\n')) {
          i--;
        }
      }
    }
  
    if(recvBufferSize < dataSize) {
      dataSize = recvBufferSize;
      if(enableDebug) {
        debugStream->println(F("SIM800L : doGet() - Buffer overflow while loading data from HTTP. Keep only first bytes..."));
      }
    }
  
    // We are expecting a final OK
    if(!readResponseCheckAnswer_P(DEFAULT_TIMEOUT, AT_RSP_OK)) {
      if(enableDebug) debugStream->println(F("SIM800L : doGet() - Invalid end of data while reading HTTP result from the module"));
      return 705;
    }
  
    if(enableDebug) {
      debugStream->print(F("SIM800L : doGet() - Received from HTTP GET : "));
      debugStream->println(recvBuffer);
    }
  }
*/
```

SciWax, 2020-07-24:
Today, I've tried out your HTTPS_POST_SoftSerial.ino version to see, if this example would work flawless. Unfortunately I've got the same issue here. 

Here the only changed parts of the code:

```
const char APN[] = "internet.telekom";
const char URL[] = "https://url.com/webhook_addv2.php";

const char CONTENT_TYPE[] = "application/json";
const char PAYLOAD[] = "{\"temperature\": \"14.11\", \"humidity\": \"3.44\", \"luminosity\": \"1.44\"}";
```
Again, the custom url is 36 characters long. This link is basically a tunnel I've created via dataplicity.io to connect to my database on my Raspberry Pi. The link looks kinda like this (Didn't post the real one for obvious reasons):

``https://dusiticks-maria-1234.dataplicity.io
``
I don't think that's the reason though and I try to give you as much information as I can.

The stuck output looks like this:

```
19:38:21.732 -> Problem to initialize AT command, retry in 1 sec
19:38:22.752 -> Setup Complete!
19:38:24.855 -> Signal OK (strenght: 14)
19:38:26.932 -> Network registration OK
19:38:28.087 -> GPRS config OK
19:38:29.716 -> GPRS connected !
19:38:29.716 -> Start HTTP POST...

```
The data gets successfully posted to my database, but then the SIM800L gets stuck again and keeps blinking very fast, because it's still in GPRS mode.

ostaquet, 2020-07-27:
@SciWax As discussed in #17, did you check the available memory? You are maybe experiencing a buffer overflow due to additional headers added by dataplicity.io.

SciWax, 2020-07-27:
> @SciWax As discussed in #17, did you check the available memory? You are maybe experiencing a buffer overflow due to additional headers added by dataplicity.io.

I'm using a Mega2560.  Good call with dataplicity.io. Maybe the site is flooding me with too many headers. I will try to find another site, where I can dump my measurements. Can I use Postman to use GET or POST on the same huge amount of variables? I will inform you, if it worked with Postman,

ostaquet, 2020-07-28:
The Mega 2560 has 8KB of RAM. It should be enough to manage the transfer but it depends on your code and how you managed the memory. Memory is a scarce resource on microcontrollers ;-)

Check the memory used after the compilation + run some [freemem routines](https://playground.arduino.cc/Code/AvailableMemory/) in the debug to check if you are overflowing the memory.

closed 2020-09-25 by ostaquet