ostaquet/Arduino-SIM800L-driver #18: HTTP Post SIM800L + Arduino Mega + Server Node JS error 408

TheLarre, 2020-07-28:
Hello ostaquet,

Im having a problem with ur driver, all the time i have an error 408 when i try to Post to my server dat its located into my laptop, i configure the server and my router and i can get the url outside my network, i try with my phone and with other laptop with postman, all work fine, but when i used sim800l all the time i have an error 408.

This is my code:

`#include <SoftwareSerial.h>
/*SIM 800L**************************************************************/
#include "SIM800L.h"
#define SIM800_TX_PIN 11
#define SIM800_RX_PIN 10
#define SIM800_RST_PIN 6
static const int ledPINSIM800 = 52;
static const char APN[] = "em";
static const char URL[] = "https//188.76.89.36:7313/getSIM800L";
static const char CONTENT_TYPE[] = "application/json";
const char PAYLOAD[] = "{\"name\": \"morpheus\", \"job\": \"leader\"}";
SIM800L *sim800l;

void setup()
{
  // Initialize Serial Monitor for debugging
  Serial.begin(9600);
  while (!Serial)
    ;
  // Setup module for SIM800 communication
   setupSIM800();
  
  // Setup module for DHT
  //setupDHT();
  // Setup module for GPS
  //setupGPS();
}
void loop()
{

  if(isGPRSconnect(false)){
Serial.println("GPRS conectado");
  Serial.println("Esperamos 5 segundos");
  delay(5000);
  Serial.println("Hecho");
  Serial.println("Esperamos 20 segundos post");
  postData();
  Serial.println("20 segundos pasados");
  resetSIM();
  }
  
}
void setupSIM800()
{
  //pinMode(ledPINSIM800 , OUTPUT);
  // Wait until the module is ready to accept AT commands
  initSim800();
  // Wait for the GSM signal
  initSignal();
  // Wait for operator network registration (national or roaming network)
  initOpertatorNetwork();
  // Setup APN for GPRS configuration
  initGPRS();
  //digitalWrite(ledPINSIM800 , HIGH);   // poner el Pin en HIGH
}
void initSim800()
{
  // Initialize a SoftwareSerial
  SoftwareSerial *serial = new SoftwareSerial(SIM800_TX_PIN, SIM800_RX_PIN);
  serial->begin(9600);
  delay(1000);
  // Initialize SIM800L driver with an internal buffer of 200 bytes and a reception buffer of 512 bytes, debug disabled
  sim800l = new SIM800L((Stream *)serial, SIM800_RST_PIN, 1024, 1024);
  sim800l->reset();
  while (!sim800l->isReady())
  {
    Serial.println(F("Problem to initialize AT command, retry in 1 sec"));
    delay(1000);
  }
  Serial.println(F("Setup Complete!"));
}
void initSignal()
{
  uint8_t signal = sim800l->getSignal();
  while (signal <= 0)
  {
    delay(1000);
    signal = sim800l->getSignal();
  }
  Serial.print(F("Signal OK (strenght: "));
  Serial.print(signal);
  Serial.println(F(")"));
  delay(1000);
}
void initOpertatorNetwork()
{
  NetworkRegistration network = sim800l->getRegistrationStatus();
  while (network != REGISTERED_HOME && network != REGISTERED_ROAMING )
  {
    delay(5000);
    network = sim800l->getRegistrationStatus();
  }
  Serial.print(F("NETWORK OK"));
  

}
void initGPRS()
{
  bool success = sim800l->setupGPRS(APN);
  while (!success)
  {
    success = sim800l->setupGPRS(APN);
    delay(5000);
  }
  Serial.println(F("GPRS config OK"));
}
bool isGPRSconnect(bool connected)
{
  for (uint8_t i = 0; i < 10 && !connected; i++)
  {
    delay(1000);
    connected = sim800l->connectGPRS();
  }
  return connected;
}
void startDisconnectGPRS(bool connected)
{
  bool disconnected = sim800l->disconnectGPRS();
  for (uint8_t i = 0; i < 5 && !connected; i++)
  {
    delay(1000);
    disconnected = sim800l->disconnectGPRS();
  }

  if (disconnected)
  {
    Serial.println(F("GPRS disconnected !"));
  }
  else
  {
    Serial.println(F("GPRS still connected !"));
  }
}
void postData()
{
  Serial.println(F("Start HTTP POST..."));
  uint16_t rc = sim800l->doPost(URL, CONTENT_TYPE, PAYLOAD, 20000, 20000);
  if (rc == 200)
  {
    // Success, output the data received on the serial
    Serial.print(F("HTTP POST successful ("));
    Serial.print(sim800l->getDataSizeReceived());
    Serial.println(F(" bytes)"));
    Serial.print(F("Received : "));
    Serial.println(sim800l->getDataReceived());
  }
  else
  {
    // Failed...
    Serial.print(F("HTTP POST error "));
    Serial.println(rc);
  }
}
void resetSIM()
{
    sim800l->reset();
    delay(5000);
    setupSIM800();

}
void goLowModeSIM()
{
  sim800l->setPowerMode(MINIMUM);
}
void goHighModeSIM(){
  sim800l->setPowerMode(NORMAL);
}`

TheLarre, 2020-07-28:
`bool isGPRSconnect(bool connected)
{
  for (uint8_t i = 0; i < 10 && !connected; i++)
  {
    delay(1000);
    connected = sim800l->connectGPRS();
  }
  return connected;
}`

TheLarre, 2020-07-28:
The SIM is a multinetwork SIM

TheLarre, 2020-07-28:
13:26:56.464 -> Start HTTP POST...
13:26:56.510 -> SIM800L : Send "AT+HTTPINIT"
13:26:56.510 -> SIM800L : End of transmission
13:26:56.556 -> SIM800L : Receive "AT+HTTPINIT

13:26:56.602 -> OK
13:26:56.602 -> "
13:26:56.602 -> SIM800L : Send "AT+HTTPPARA="CID",1"
13:26:56.650 -> SIM800L : End of transmission
13:26:56.650 -> SIM800L : Receive "AT+HTTPPARA="CID",1

13:26:56.698 -> OK
13:26:56.698 -> "
13:26:56.698 -> SIM800L : Send "AT+HTTPPARA="URL","https//188.76.89.36:7313/getSIM800L""
13:26:56.837 -> SIM800L : End of transmission
13:26:56.883 -> SIM800L : Receive "AT+HTTPPARA="URL","https//188.76.89.36:7313/getSIM800L"

13:26:56.977 -> OK
13:26:56.977 -> "
13:26:56.977 -> SIM800L : Send "ATI"
13:26:56.977 -> SIM800L : End of transmission
13:26:57.023 -> SIM800L : Receive "ATI

13:26:57.069 -> SIM800 R14.18
13:26:57.069 -> "
13:26:57.069 -> SIM800L : initiateHTTP() - Support of SSL enabled
13:26:57.117 -> SIM800L : Send "AT+HTTPSSL=0"
13:26:57.163 -> SIM800L : End of transmission
13:26:57.209 -> SIM800L : Receive "AT+HTTPSSL=0

13:26:57.209 -> OK
13:26:57.209 -> "
13:26:57.255 -> SIM800L : Send "AT+HTTPPARA="CONTENT","application/json""
13:26:57.348 -> SIM800L : End of transmission
13:26:57.348 -> SIM800L : Receive "AT+HTTPPARA="CONTENT","application/json"

13:26:57.440 -> OK
13:26:57.440 -> "
13:26:57.440 -> SIM800L : Send "AT+HTTPDATA=37,20000"
13:26:57.486 -> SIM800L : End of transmission
13:26:57.486 -> SIM800L : Receive "AT+HTTPDATA=37,20000

13:26:57.533 -> DOWNLOAD
13:26:57.533 -> "
13:26:57.579 -> SIM800L : doPost() - Payload to send : {"name": "morpheus", "job": "leader"}
13:26:58.091 -> SIM800L : Send "AT+HTTPACTION=1"
13:26:58.137 -> SIM800L : End of transmission
13:26:58.184 -> SIM800L : Receive "AT+HTTPACTION=1

13:26:58.231 -> OK
13:26:58.231 -> "
13:27:18.148 -> SIM800L : Receive timeout
13:27:18.195 -> SIM800L : doPost() - Server timeout
13:27:18.242 -> HTTP POST error 408

TheLarre, 2020-07-28:
[image]
[image]

TheLarre, 2020-07-28:
Okay i just try to do a GET petition with your example and i see dat u reset the sim when u init, so in my code i reset it again, this can be one problem dat i solved right now but either if i use ur post softwareserial example i have the same error aswell

ostaquet, 2020-07-29:
@TheLarre Can you post the full debug output? While reading the code, I have the feeling you go multiple time in some routine that could impact the behaviour of the SIM800.

TheLarre, 2020-07-29:
09:48:51.333 -> SIM800L : Reset
09:48:53.343 -> SIM800L : Prepare internal buffer of 1024 bytes
09:48:53.391 -> SIM800L : Prepare reception buffer of 1024 bytes
09:48:53.437 -> SIM800L : Send "AT"
09:48:58.386 -> SIM800L : Receive timeout
09:48:58.433 -> Problem to initialize AT command, retry in 1 sec
09:48:59.412 -> SIM800L : Send "AT"
09:48:59.460 -> SIM800L : End of transmission
09:48:59.460 -> SIM800L : Receive "AT

09:48:59.506 -> OK
09:48:59.506 -> "
09:48:59.506 -> Setup Complete!
09:48:59.506 -> SIM800L : Send "AT+CSQ"
09:48:59.553 -> SIM800L : End of transmission
09:48:59.553 -> SIM800L : Receive "AT+CSQ

09:48:59.600 -> +CSQ: 0,0
09:48:59.600 -> "
09:49:00.529 -> SIM800L : Send "AT+CSQ"
09:49:00.575 -> SIM800L : End of transmission
09:49:00.623 -> SIM800L : Receive "AT+CSQ

09:49:00.669 -> +CSQ: 0,0
09:49:00.669 -> "
09:49:01.603 -> SIM800L : Send "AT+CSQ"
09:49:01.649 -> SIM800L : End of transmission
09:49:01.649 -> SIM800L : Receive "AT+CSQ

09:49:01.696 -> +CSQ: 22,0
09:49:01.696 -> "
09:49:01.696 -> Signal OK (strenght: 22)
09:49:02.674 -> SIM800L : Send "AT+CREG?"
09:49:02.721 -> SIM800L : End of transmission
09:49:02.721 -> SIM800L : Receive "AT+CREG?

09:49:02.768 -> +CREG: 0,2
09:49:02.768 -> "
09:49:07.701 -> SIM800L : Send "AT+CREG?"
09:49:07.747 -> SIM800L : End of transmission
09:49:07.795 -> SIM800L : Receive "AT+CREG?

09:49:07.795 -> +CREG: 0,5
09:49:07.841 -> "
09:49:07.841 -> NETWORK OKSIM800L : Send "AT+SAPBR=3,1,"Contype","GPRS""
09:49:07.887 -> SIM800L : End of transmission
09:49:07.934 -> SIM800L : Receive "AT+SAPBR=3,1,"Contype","GPRS"

09:49:07.981 -> OK
09:49:07.981 -> "
09:49:07.981 -> SIM800L : Send "AT+SAPBR=3,1,"APN","em""
09:49:08.027 -> SIM800L : End of transmission
09:49:08.075 -> SIM800L : Receive "AT+SAPBR=3,1,"APN","em"

09:49:08.123 -> OK
09:49:08.123 -> "
09:49:08.123 -> GPRS config OK
09:49:09.055 -> SIM800L : Send "AT+SAPBR=1,1"
09:49:09.662 -> SIM800L : End of transmission
09:49:09.708 -> SIM800L : Receive "AT+SAPBR=1,1

09:49:09.755 -> OK
09:49:09.755 -> "
09:49:09.755 -> GPRS conectado
09:49:09.755 -> Esperamos 5 segundos
09:49:14.696 -> Hecho
09:49:14.743 -> Esperamos 20 segundos post
09:49:14.743 -> Start HTTP POST...
09:49:14.789 -> SIM800L : Send "AT+HTTPINIT"
09:49:14.789 -> SIM800L : End of transmission
[... 289 lines cut ...]
09:52:56.432 -> OK
09:52:56.432 -> "
09:53:06.362 -> SIM800L : Receive timeout
09:53:06.408 -> SIM800L : doPost() - Server timeout
09:53:06.455 -> HTTP POST error 408
09:53:06.455 -> SIM800L : Send "AT+SAPBR=0,1"
09:53:06.501 -> SIM800L : End of transmission
09:53:06.548 -> SIM800L : Receive "AT+SAPBR=0,1

09:53:06.548 -> ERROR
09:53:06.595 -> "
09:53:06.595 -> GPRS still connected !
09:53:06.595 -> SIM800L : Send "AT+CFUN?"
09:53:06.642 -> SIM800L : End of transmission
09:53:06.642 -> SIM800L : Receive "AT+CFUN?

09:53:06.688 -> +CFUN: 1
09:53:06.688 -> "
09:53:06.688 -> SIM800L : Send "AT+CFUN=0"
09:53:06.735 -> SIM800L : Send "AT+CFUN?"
09:53:06.970 -> SIM800L : End of transmission
09:53:07.017 -> SIM800L : Receive "
09:53:07.017 -> +CPIN: NOT READY
09:53:07.064 -> "
09:53:07.064 -> Failed to switch module to low power mode

TheLarre, 2020-08-02:
anyone has this problem too??

ostaquet, 2020-08-04:
It seems that you have a read timeout during the reading of the data back.

How to you power the SIM800L?
Do you have some logs on the server side? (according to the log on SIM800L side, the connection was established but the server didn't send any data)

TheLarre, 2020-08-04:
The sim800l is powered by a voltage convertor, the sim800l have a 4.1V.

If I have a request on the server I would see it by console, but I see nothing when I make a post with sim800l but with postman I see that request, the server sends a true or false

ostaquet, 2020-08-04:
On the power supply, could you power the SIMM800L module with an external DC power supply? (the consumption is quite high during transmission and the voltage convertor may not supply enough current).

Additionally, I'm surprised of the `+CPIN: NOT READY` in one of your log. Some research on Internet shows that it could be lead from multiple issues ([see this](https://community.hologram.io/t/random-no-carrier-with-sim800-sim900-modems-adafruit-fona-on-raspberry-pi/244/10))

As you didn't see any TCP connection on the server side, we should assume that the SIM800L didn't connect to the internet.

TheLarre, 2020-08-04:
It may be a matter of the voltage, I am going to try to connect 5V to the sim800L, but it sends data from time to time because  I can visualize the data expense of the sim

TheLarre, 2020-08-04:
Yep 5V needed it works

closed 2020-08-04 by ostaquet