ostaquet/Arduino-SIM800L-driver #19: How to comunicate with modul manualy ?

bebeno, 2020-08-03:
Hi Olivier.
I would like to ask you one question. I want use your library for GPRS communication because for me is this library the best but I need read and send SMS as well.
I am trying comunicate manualy with modul over you library but I do not know how to send AT command and then read answer from buffer. Can you point me please?

Thanks

ostaquet, 2020-08-04:
You can extend the code by sending command with the method: 
`void sendCommand(const char* command)`

The CRLF is added by the method, so, you just have to use the function (by example):
`sendComment("AT+GMR")`

To read the response from the module, you can use the method:
`bool readResponse(uint16_t timeout, uint8_t crlfToWait)`

The return value is true if there is a result, false elsewhere.
The `timeout` is the timeout defined in milliseconds to have the results and `crlfToWait` is the number of CRLF to ignore before catching the response (quite handy as the module generates CRLF depending on the command.

If the return value is true, there is some data in the internal buffer (`char *internalBuffer`).

All those methods and variables are in private. You should put them in protected before extending the class to access to low level methods.

closed 2020-08-04 by ostaquet

bebeno, 2020-08-04:
Thank you for your time and answer.
But It does not work.
I added only sendComment("AT+GMR");  into working code and I got error after compiltion. 
Maybe I did not understand something.

My error is : 

Multiple libraries were found for "SIM800L.h"
 Used: C:\Users\Ivan\Documents\Arduino\libraries\SIM800L_HTTP_connector
Multiple libraries were found for "SoftwareSerial.h"
 Used: C:\Users\Ivan\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\libraries\SoftwareSerial
 Not used: C:\Users\Ivan\Documents\Arduino\libraries\espsoftwareserial-master
exit status 1
'sendComment' was not declared in this scope

Thanks for help.

ostaquet, 2020-08-04:
Damned, the automatic corrector changed my word. The example is `sendCommand("AT+GMR");` and not `sendComment("AT+GMR");`.

If you have multiple libraries, it is because you have multiple libraries in the link edit step. Go to some forums to fix it ;-).

bebeno, 2020-08-04:
Hi.
I am very sorry that I waste your time but result is still the same.

   sendCommand("AT+GMR")ô

   ^~~~~~~~~~~

Multiple libraries were found for "SoftwareSerial.h"
 Used: C:\Users\Ivan\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\libraries\SoftwareSerial
 Not used: C:\Users\Ivan\Documents\Arduino\libraries\espsoftwareserial-master
Multiple libraries were found for "SIM800L.h"
 Used: C:\Users\Ivan\Documents\Arduino\libraries\Arduino-SIM800L-driver-master
exit status 1
'sendCommand' was not declared in this scope

ostaquet, 2020-09-25:
Seems you have multiple declaration of your libraries. 

This is not an issue with the library, it is an issue in your install. 

You can add some debug information to understand what's going on by enabling the verbose output during the compilation (can be set in the preferences of the Arduino IDE).

You can try to remove manually installed library and reinstall through the library manager, it should solve the issue.

Notice that there are forums where people will help you.