ostaquet/Arduino-SIM800L-driver #21: Software serial wrong delcaration?

TheEngineersCorner, 2020-08-27:
hello, the basic test for software serial is incorrect? the rx and tx is wrong way.

SoftwareSerial(SIM800_TX_PIN, SIM800_RX_PIN);
The correct method is:
SoftwareSerial(RX,TX)

See, https://www.arduino.cc/en/Tutorial/SoftwareSerialExample

mohammadfatemieh, 2020-09-03:
It's not different because the RX pin and TX pin is defined

#define SIM800_TX_PIN 3
#define SIM800_RX_PIN 2
#define SIM800_RST_PIN 6

mohammadfatemieh, 2020-09-03:
I think You say correctly

ostaquet, 2020-09-25:
You right, to me clean. I should invert the names.

But at the end, the TX pin of the SIM800 is connected on the RX pin of the Arduino and the RX pin of the SIM800 is connected on the TX pin of the Arduino :-P

Fixed in master branch.

Thanks for the feedback

closed 2020-09-25 by ostaquet