ostaquet/Arduino-SIM800L-driver #30: Inherit from Client class

enwi, 2020-11-25:
I was wondering if it was possible to make `SIM800L` inherit from the Arduino `Client` class. That way it would be possible to use [SSLClient](https://github.com/OPEnSLab-OSU/SSLClient) to have a more secure SSL implementation.

I have not done any project with the SIM800L yet, but was wondering if you @ostaquet think it would be feasible to do something like this?

If yes, I would be happy to help implement this 'feature'

ostaquet, 2020-11-29:
Hi @enwi,

It is a great idea to benefit of other libraries like MQTT or SSLClient.

I checked the doc and it is probably doable via the TCP and UDP connection (in the data sheet, starting page 220; AT Commands for TCPIP Application Toolkit).

Did you already look at it?

KR,
Olivier

enwi, 2020-11-29:
Hey @ostaquet I did not look at it and am already using the TinyGSM library, which already  supports that feature, since their client class inherits from Arduino `Client`. 

I also moved to the other lib, because I needed support for setting the SIM PIN. I tried implementing it in your library, but always got an error from the SIM800L, whenever I send the command, though using an example sketch which pipes the data between Serial and Serial1 and sending the commands myself, it worked. So no idea, why it would not work with:
```c++
const char AT_CMD_CPIN_TEST[] PROGMEM = "AT+CPIN?";                           // Check if PIN is needed
const char AT_CMD_CPIN[] PROGMEM = "AT+CPIN=";                                // Set the PIN

const char AT_RSP_CPIN_SIM_PIN[] PROGMEM = "+CPIN: SIM PIN";                  // Expected answer CPIN
const char AT_RSP_CPIN_READY[] PROGMEM = "+CPIN: READY";                      // Expected answer CPIN

/**
 * @brief Check if a PIN needs to be entered
 * 
 * @return true A PIN is required
 * @return false A PIN is not required
 */
bool SIM800L::needPIN() {
  sendCommand_P(AT_CMD_CPIN_TEST);
  return readResponseCheckAnswer_P(DEFAULT_TIMEOUT, AT_RSP_CPIN_SIM_PIN);
}

/**
 * @brief Set the PIN for accessing SIM card
 * 
 * @param pin The four digit pin
 * @return true If the PIN was set
 * @return false If the PIN was not set
 */
bool SIM800L::setPIN(const char *pin) {
  char cmdBuff[32];
  strcpy_P(cmdBuff, AT_CMD_CPIN);
  cmdBuff[8] = pin[0];
  cmdBuff[9] = pin[1];
  cmdBuff[10] = pin[2];
  cmdBuff[11] = pin[3];
  cmdBuff[12] = '\0';
  sendCommand(cmdBuff);
  return readResponseCheckAnswer_P(DEFAULT_TIMEOUT, AT_RSP_CPIN_READY);
}
```

joysfera, 2021-01-21:
> Hey @ostaquet I did not look at it and am already using the TinyGSM library, which already supports that feature, since their client class inherits from Arduino `Client`.

but it does not sound like the TinyGSM allows for using own SSL library because SSL/TLS works on just some of them modems and they even have to come with certain firmware revision.

enwi, 2021-01-21:
@joysfera SSL/TSL is just another layer on top of TCP:
[image]

So using [SSLClient](https://github.com/OPEnSLab-OSU/SSLClient) with TinyGSM gives you SSL/TLS on all modems that allow TCP connections (pretty much every single one), because the encryption is not done in hardware on the modem, but in software on the mcu.

joysfera, 2021-01-21:
@enwi , thank you, that helps me a lot.

closed 2021-12-29 by ostaquet