ostaquet/Arduino-SIM800L-driver #33: HTTP GET error 606

imranrajjad, 2021-01-01:
I am running the HTTPS_GET_HardwareSerial with ESP32 TTGO board which has SIM800L module integrated.

Strangely the example works fine but I try another URL such as 

`https://jsonplaceholder.typicode.com/todos/1
`
I get the error

`HTTP GET error 606`

ostaquet, 2021-01-17:
The embedded IP stack of the SIM800L only supports SSL2, SSL3 and TLS 1.0. These cryptographic protocols are considered deprecated for most of web browsers and the connection will be denied by modern backend (i.e. AWS). This will typically lead to an error `605` or `606` when you establish an HTTPS connection.

closed 2021-01-17 by ostaquet

cffiebigc, 2021-06-20:
Is it possible to update SIM800L Firmware to solve this?

ostaquet, 2021-06-23:
Unfortunately, the last version available from SIMCOM is outdated (January 2016... an eternity in the cryptographic protocols 😞 ). The firmware can be downloaded on the [SIMCOM website](https://simcom.ee/documents/?dir=SIM800L) (as you can see by visiting the website, they have some troubles with the SSL certificates 😅 ).

The code of the firmware is not available. So, the only way is to work with the SIM800L at a lower level (TCP) and implement the HTTP/SSL stack (or use a library like [WolfSSL](https://www.wolfssl.com)).

EyitopeIO, 2021-07-26:
If anyone reading this comment needs to at least test their HTTPS application with the SIM800, you can find free test servers on the internet. An example would be http://httpbin.org/anything . As at the time of this comment, this link still supports the SIM800 (I used SIM808) TLS/SSL stack.

Enisco, 2021-12-21:
The SIM800L only supports up to TLS 1.0. There are still some APIs that still support TLS1.0 or even lower SSLs, I still used it for Heroku last month. However, I recently hosted an API on digitalOcean but unfortunately, the SSL incompatibility issue arises and I'm unable to successfully post data. A solution is to manually downgrade the API to TLS 1.0 compatible or use a third-party SSL library like PPPoS

Overite, 2023-02-01:
> Unfortunately, the last version available from SIMCOM is outdated (January 2016... an eternity in the cryptographic protocols 😞 ). The firmware can be downloaded on the [SIMCOM [...]

please how can i use this wolfssl on my sim800 project

Overite, 2023-02-01:
i really nee a solution on this guys please help me out