ostaquet/Arduino-SIM800L-driver #34: sim800l->doPost Limit Size

lasida, 2021-01-06:
Hi. 

i use this library and it works fine. the function of sim800l->doPost is using data type char for sending the payload, and in esp32 limit char is 4000bytes, can I send 25.000 bytes of data json in one post, or I must split the json to post the data,

I was using String type data for handle the 25.000 bytes of data, but it can't fit to functions sim800l->doPost.

maybe you have a suggestion
Thanks :D

bhupiister, 2021-01-07:
@lasida any luck? I want to send log file via post method. Do you think it's possible?

lasida, 2021-01-07:
yes, it's possible in ESP32, I sending 21100bytes, or 21.1 kB data of JSON with base64image string
I just create char jsonVision[30000] outside the loop before setup(). and wrapping with library ArduinoJson 

And using SpiRamAllocator for increasing JsonDocument
https://arduinojson.org/v6/how-to/use-external-ram-on-esp32/

Before I using SpiRamAllocator its just 8.8kB data received.
I think it's limit JsonDocument default from ArduinoJson Lib

closed 2021-01-15 by lasida