ostaquet/Arduino-SIM800L-driver #41: HTTP GET error 701

TexhFex, 2021-03-30:
Hello, i hope i can describe my problem well enough:

I use the example HTTP GET script to write data to my website (PHP $_GET ...). Every time i start my script the first pass is successful with Error 705 (it sends the data and does not hand back a site). But now on the next passes i get the Error 701. 
My program:
```
...
void setup(){
   setupModul();
}

void loop(){
messure data ...
setPowerMode NORMAL ...
GPRS connect ...
doGET(xyz) ...
setPowerMode MINIMUM;
wait for 30 seconds ...
}
```
Did I forget to activate something or do I have to clear the cache somehow?
I hope you understand my problem. 
Thanks in advance

Felix

rajdeep-engineer, 2021-03-31:
I am facing the similar issues with both GET and POST requests: Errors 703, 701.

TexhFex, 2021-03-31:
I found this in the cpp file: 
```
sendCommand_P(AT_CMD_HTTPINIT);
  if(!readResponseCheckAnswer_P(DEFAULT_TIMEOUT, AT_RSP_OK)) {
    if(enableDebug) debugStream->println(F("**SIM800L : initiateHTTP() - Unable to init HTTP**"));
    return 701;
  }
```

Has somebody an idea how to fix it?

TexhFex, 2021-03-31:
This are the typical error codes:
701 Alert state 
702 Unknown error
703 Busy

rajdeep-engineer, 2021-03-31:
Ok I have gone through the source code and figured out the specific issue bothering me:-

Although my module version is >14, I would like to communicate with my cloud server using HTTP:// protocol and not HTTPS://. Because if I choose HTTPS:// my server will expect my client to follow TLS 1.3 which is not possible on the outdated SIM800L module I am using.

But the piece of code below doesn't address usage of HTTP:// protocol on module version >14
[image]

TexhFex, 2021-04-01:
I am using a own programmed API to transfer the data. 
Is it possible to clear the cache of the last doGet() to "reset" the modul but not to disconnect from the network?

TexhFex, 2021-04-04:
When I give the module direct the AT commands via serial monitor, I am getting an **HTTP error code 601** after the first time starting the http call (this works fine HTTP 200). Terminating the HTTP process and starting a new one ends up with the same error ...

_Could this be a error of the SIM800L module?_

ostaquet, 2021-05-22:
Would you add the full debug log with the AT command and answers from the module?
It will probably give some paths to investigate the issue.

TexhFex, 2021-05-23:
Thanks for answering.

Basicly my SIM800L modules comes out with this error when i try to send more than one a HTTP GET. 

I fixed it by the software reset of the library, but so it have to connect by every HTTP GET again.

I can´t send you the debug log because it don´t have it anymore. 

If you have an idea, i could try to test the program and generate a debug log. 

Thanks in advance.

ostaquet, 2021-06-26:
It is probably an issue in the main function in your code.

Just tested multiple HTTPS GET without any issues:
```
17:47:40.515 -> Problem to initialize AT command, retry in 1 sec
17:47:41.530 -> Setup Complete!
17:47:43.632 -> Signal OK (strenght: 17)
17:47:45.731 -> Network registration OK
17:47:46.868 -> GPRS config OK
17:47:48.166 -> GPRS connected !
17:47:48.166 -> Start first HTTP GET...
17:47:53.696 -> HTTP GET successful (291 bytes)
17:47:53.696 -> Received : {"args":{"foo1":"bar1","foo2":"bar2"},"headers":{"x-forwarded-proto":"https","x-forwarded-port":"443","host":"postman-echo.com","x-amzn-trace-id":"Root=1-60d74c28-0c51c1db2f416ad360bab791","accept":"*/*","user-agent":"SIMCOM_MODULE"},"url":"https://postman-echo.com/get?foo1=bar1&foo2=bar2"}
17:47:53.730 -> Start second HTTP GET...
17:47:55.943 -> HTTP GET successful (291 bytes)
17:47:55.943 -> Received : {"args":{"foo1":"bar1","foo2":"bar2"},"headers":{"x-forwarded-proto":"https","x-forwarded-port":"443","host":"postman-echo.com","x-amzn-trace-id":"Root=1-60d74c2b-5ed389a33798851612687079","accept":"*/*","user-agent":"SIMCOM_MODULE"},"url":"https://postman-echo.com/get?foo1=bar1&foo2=bar2"}
17:47:55.979 -> Start third HTTP GET...
17:47:58.318 -> HTTP GET successful (291 bytes)
17:47:58.318 -> Received : {"args":{"foo1":"bar1","foo2":"bar2"},"headers":{"x-forwarded-proto":"https","x-forwarded-port":"443","host":"postman-echo.com","x-amzn-trace-id":"Root=1-60d74c2d-4274878e1cabc2d34e482c82","accept":"*/*","user-agent":"SIMCOM_MODULE"},"url":"https://postman-echo.com/get?foo1=bar1&foo2=bar2"}
17:47:58.561 -> GPRS disconnected !
17:47:58.634 -> Module in low power mode
```

closed 2021-06-26 by ostaquet