ostaquet/Arduino-SIM800L-driver #48: How to construct correct HTTP Headers for SIM800L-driver library

mitiay7, 2021-09-12:
Dear Sir, 

Please advise ho to correctly construct headers which are consist from more than one parametr?

I'm trying to make GET request with two auth headers (X-Parse-Application-Id and X-Parse-REST-API-Key) to PARSE backend but getting 702 error. If I switch on extensive logging - getting the following error

13:46:58.916 -> SIM800L : initiateHTTP() - Unable to define Headers
13:46:58.916 -> HTTP GET error 702

ostaquet, 2021-11-04:
Hi @mitiay7,

The header part has been implemented by @ztaras and I never had the need to use header in my applications.

In the pull request (https://github.com/ostaquet/Arduino-SIM800L-driver/pull/25#issuecomment-725900615), there is some insight about the construction of the headers; quoted below.

> This should do the trick:
```sim800l->doGet("https://postman-echo.com/get?foo1=bar1&foo2=bar2", "Authorization:Basic aW50ZWdy...", 10000);```

> If you also want to send the Content-Type header, the line would look like this:
```sim800l->doGet("https://postman-echo.com/get?foo1=bar1&foo2=bar2", "Authorization:Basic aW50ZWdy...\\r\\nContent-Type:application/json", 10000);```

> Just replace the ```aW50ZWdy...``` part with whatever Postman generated for your username and password.

ztaras, 2021-11-04:
Dmitry, 
Please try using double back slash in between two headers, e.g.:
`sim800l->doGet("https://postman-echo.com/get?foo1=bar1&foo2=bar2", "Header-1:value1\\r\\nHeader-2:value2", 10000);`

EndruK, 2022-02-28:
I figured out that it is connected to the length of the char array - an array with more than 34 characters results in getting stuck. With <= 34 characters it works. 
Could this be connected to message buffer sizes?

ostaquet, 2022-03-02:
I didn't see any buffer limits in this part of the code.

The message buffer is defined at the initialisation of the driver (constructor `SIM800L::SIM800L(...)`). What is the message buffer size you are using?

Can you enable debug mode to see the command going in and out of the module?

EndruK, 2022-03-10:
Ok after some research I finally figured out a solution.
As described [here](https://arduino.stackexchange.com/questions/37221/common-ground-serial-communication-trouble-sim800l) the SIM800L module seems to have issues with automatic baud detection.
Setting the baud rate manually solves all communication issues - even sending longer headers :)
```
AT+IPR=9600
AT&W
```

ostaquet, 2024-06-02:
As mentioned in the README, the project is no longer maintained as my module is broken. Thanks for reporting the issue.

closed 2024-06-02 by ostaquet