ostaquet/Arduino-SIM800L-driver #49: Problem of reading and processing a Json string

SeverinBaumgartner, 2021-09-17:
Hi there,

I want to get a json string, for example one from Openweathermap.
it works perfectly, but then I want to access parameters from the string with the ArduinoJson libary in order to use them further there is some trouble.

THE PROBLEM
I am almost certain that the problem is with the delivery of: 

`String input = sim800l->getDataReceived();`

I would be very happy to receive suggestions for solutions.

greetings from Germany!

`````#include <SoftwareSerial.h>

#include "SIM800L.h"

#include <stdio.h>
#include <stdlib.h>

#include <ArduinoJson.h>
#include <Arduino.h>

#define SIM800_RX_PIN 7
#define SIM800_TX_PIN 8
#define SIM800_RST_PIN 6

const char APN[] = "internet.eplus.de";
const char URL[] = "http://api.openweathermap.org/data/2.5/weather?id=2843729&appid=6fe64bb0ac0ac7504c1a2e56672416d4";

SIM800L* sim800l;

void setup() {  
  // Initialize Serial Monitor for debugging
  Serial.begin(9600);
  while(!Serial);

  // Initialize a SoftwareSerial
  SoftwareSerial* serial = new SoftwareSerial(SIM800_RX_PIN, SIM800_TX_PIN);
  serial->begin(9600);
  delay(1000);
   
  // Initialize SIM800L driver with an internal buffer of 200 bytes and a reception buffer of 512 bytes, debug disabled
  sim800l = new SIM800L((Stream *)serial, SIM800_RST_PIN, 200, 600);

  // Equivalent line with the debug enabled on the Serial
  // sim800l = new SIM800L((Stream *)serial, SIM800_RST_PIN, 200, 512, (Stream *)&Serial);

  // Setup module for GPRS communication
  setupModule();
}
 
void loop() 

{
  // Establish GPRS connectivity (5 trials)
  bool connected = false;
  for(uint8_t i = 0; i < 5 && !connected; i++) {
    delay(1000);
    connected = sim800l->connectGPRS();
  }

  // Check if connected, if not reset the module and setup the config again
  if(connected) {
    Serial.print(F("GPRS connected with IP "));
    Serial.println(sim800l->getIP());
  } else {
    Serial.println(F("GPRS not connected !"));
    Serial.println(F("Reset the module."));
    sim800l->reset();
    setupModule();
    return;
  }

  Serial.println(F("Start HTTP GET..."));

  // Do HTTP GET communication with 10s for the timeout (read)
  uint16_t rc = sim800l->doGet(URL, 20000);
   if(rc == 200) {
    // Success, output the data received on the serial
    Serial.print(F("HTTP GET successful ("));
    Serial.print(sim800l->getDataSizeReceived());
    Serial.println(F(" bytes)"));
    Serial.print(F("Received : "));
    Serial.println(sim800l->getDataReceived());

    String input = sim800l->getDataReceived();                      
    
    DynamicJsonDocument doc(800);

    DeserializationError error = deserializeJson(doc, input);

    if (error) {
      Serial.print(F("deserializeJson() failed: "));
      Serial.println(error.f_str());
    return;
    }

   JsonObject main = doc["main"];
   float main_temp = main["temp"]; /
   float main_feels_like = main["feels_like"];   
   float main_temp_min = main["temp_min"];    
   float main_temp_max = main["temp_max"];    
   int main_pressure = main["pressure"];       
   int main_humidity = main["humidity"];     
   int main_sea_level = main["sea_level"];       
   int main_grnd_level = main["grnd_level"];      

   Serial.println(main_temp);
   
   }
 
 
  else {
    // Failed...
    Serial.print(F("HTTP GET error "));
    Serial.println(rc);
  }

  delay(1000);

  // Close GPRS connectivity (5 trials)
  bool disconnected = sim800l->disconnectGPRS();
  for(uint8_t i = 0; i < 5 && !connected; i++) {
    delay(1000);
    disconnected = sim800l->disconnectGPRS();
  }
  
  if(disconnected) {
    Serial.println(F("GPRS disconnected !"));
  } else {
    Serial.println(F("GPRS still connected !"));
  }

}

void setupModule() {
    // Wait until the module is ready to accept AT commands
  while(!sim800l->isReady()) {
    Serial.println(F("Problem to initialize AT command, retry in 1 sec"));
    delay(1000);
  }
  Serial.println(F("Setup Complete!"));

  // Wait for the GSM signal
  uint8_t signal = sim800l->getSignal();
  while(signal <= 0) {
    delay(1000);
    signal = sim800l->getSignal();
  }
  Serial.print(F("Signal OK (strenght: "));
  Serial.print(signal);
  Serial.println(F(")"));
  delay(1000);

  // Wait for operator network registration (national or roaming network)
  NetworkRegistration network = sim800l->getRegistrationStatus();
  while(network != REGISTERED_HOME && network != REGISTERED_ROAMING) {
    delay(1000);
    network = sim800l->getRegistrationStatus();
  }
  Serial.println(F("Network registration OK"));
  delay(1000);

  // Setup APN for GPRS configuration
  bool success = sim800l->setupGPRS(APN);
  while(!success) {
    success = sim800l->setupGPRS(APN);
    delay(5000);
  }
  Serial.println(F("GPRS config OK"));
}````

ostaquet, 2021-11-04:
I'm not a specialist of the ArduinoJson library.

You didn't explain what problem you are encountering. So, it is a bit difficult to give clear directions.

First, you should be sure that the reception buffer is big enough to receive the data. I see you have 600 bytes reserved in your code. Is it enough?

Second, what kind of MCU are you using? If you are running on Arduino Uno with 2KB of RAM, it will most certainly crash when you assign 600 bytes + 200 bytes for the buffers. Keep an eye on your memory usage.

Third, you are using the ```getDataReceived()``` which returns a pointer to a char array and assign it to a String. This approach has to be checked for two reasons: the ability of the library to convert char array to String and the memory area pointed by the pointer could be used in the future and evolve. I suggest starting by copying the result of ```getDataReceived()```into the main code through ```memcpy``` and print it to ensure you have a complete and reliable data in this area.

adamngaila, 2022-01-04:
> Hi there,
> 
> I want to get a json string, for example one from Openweathermap. it works perfectly, but then I want to access parameters from the string with the ArduinoJson libary in order to use them further there is some trouble.
> 
> THE PROBLEM I am almost certain that the problem is with the delivery of:
> 
> `String input = sim800l->getDataReceived();`
> 
> I would be very happy to receive suggestions for solutions.
> 
> greetings from Germany!
> 
> ```
> 
> #include "SIM800L.h"
> 
> #include <stdio.h>
> #include <stdlib.h>
> 
> 
> #include <ArduinoJson.h>
> #include <Arduino.h>
> 
> #define SIM800_RX_PIN 7
> #define SIM800_TX_PIN 8
> #define SIM800_RST_PIN 6
> 
> const char APN[] = "internet.eplus.de";
> const char URL[] = "http://api.openweathermap.org/data/2.5/weather?id=2843729&appid=6fe64bb0ac0ac7504c1a2e56672416d4";
> 
> SIM800L* sim800l;
> 
> 
> void setup() {  
>   // Initialize Serial Monitor for debugging
>   Serial.begin(9600);
>   while(!Serial);
> 
>   // Initialize a SoftwareSerial
>   SoftwareSerial* serial = new SoftwareSerial(SIM800_RX_PIN, SIM800_TX_PIN);
>   serial->begin(9600);
>   delay(1000);
>    
>   // Initialize SIM800L driver with an internal buffer of 200 bytes and a reception buffer of 512 bytes, debug disabled
>   sim800l = new SIM800L((Stream *)serial, SIM800_RST_PIN, 200, 600);
> 
>   // Equivalent line with the debug enabled on the Serial
>   // sim800l = new SIM800L((Stream *)serial, SIM800_RST_PIN, 200, 512, (Stream *)&Serial);
> 
>   // Setup module for GPRS communication
>   setupModule();
> }
>  
> void loop() 
> 
> {
>   // Establish GPRS connectivity (5 trials)
>   bool connected = false;
>   for(uint8_t i = 0; i < 5 && !connected; i++) {
>     delay(1000);
>     connected = sim800l->connectGPRS();
>   }
> 
>   // Check if connected, if not reset the module and setup the config again
>   if(connected) {
>     Serial.print(F("GPRS connected with IP "));
>     Serial.println(sim800l->getIP());
>   } else {
>     Serial.println(F("GPRS not connected !"));
>     Serial.println(F("Reset the module."));
>     sim800l->reset();
>     setupModule();
>     return;
>   }
> 
>   Serial.println(F("Start HTTP GET..."));
> 
>   // Do HTTP GET communication with 10s for the timeout (read)
>   uint16_t rc = sim800l->doGet(URL, 20000);
>    if(rc == 200) {
>     // Success, output the data received on the serial
>     Serial.print(F("HTTP GET successful ("));
>     Serial.print(sim800l->getDataSizeReceived());
>     Serial.println(F(" bytes)"));
>     Serial.print(F("Received : "));
>     Serial.println(sim800l->getDataReceived());
> 
>     String input = sim800l->getDataReceived();                      
>     
>     DynamicJsonDocument doc(800);
> 
>     DeserializationError error = deserializeJson(doc, input);
> 
>     if (error) {
>       Serial.print(F("deserializeJson() failed: "));
>       Serial.println(error.f_str());
>     return;
>     }
> 
>    JsonObject main = doc["main"];
>    float main_temp = main["temp"]; /
>    float main_feels_like = main["feels_like"];   
>    float main_temp_min = main["temp_min"];    
>    float main_temp_max = main["temp_max"];    
>    int main_pressure = main["pressure"];       
>    int main_humidity = main["humidity"];     
>    int main_sea_level = main["sea_level"];       
>    int main_grnd_level = main["grnd_level"];      
> 
>    Serial.println(main_temp);
>    
>    }
>  
>  
>   else {
>     // Failed...
>     Serial.print(F("HTTP GET error "));
>     Serial.println(rc);
>   }
> 
>   delay(1000);
> 
>   // Close GPRS connectivity (5 trials)
>   bool disconnected = sim800l->disconnectGPRS();
>   for(uint8_t i = 0; i < 5 && !connected; i++) {
>     delay(1000);
>     disconnected = sim800l->disconnectGPRS();
>   }
>   
>   if(disconnected) {
>     Serial.println(F("GPRS disconnected !"));
>   } else {
>     Serial.println(F("GPRS still connected !"));
>   }
> 
> }
> 
> void setupModule() {
>     // Wait until the module is ready to accept AT commands
>   while(!sim800l->isReady()) {
>     Serial.println(F("Problem to initialize AT command, retry in 1 sec"));
>     delay(1000);
>   }
>   Serial.println(F("Setup Complete!"));
> 
>   // Wait for the GSM signal
>   uint8_t signal = sim800l->getSignal();
>   while(signal <= 0) {
>     delay(1000);
>     signal = sim800l->getSignal();
>   }
>   Serial.print(F("Signal OK (strenght: "));
>   Serial.print(signal);
>   Serial.println(F(")"));
>   delay(1000);
> 
>   // Wait for operator network registration (national or roaming network)
>   NetworkRegistration network = sim800l->getRegistrationStatus();
>   while(network != REGISTERED_HOME && network != REGISTERED_ROAMING) {
>     delay(1000);
>     network = sim800l->getRegistrationStatus();
>   }
>   Serial.println(F("Network registration OK"));
>   delay(1000);
> 
>   // Setup APN for GPRS configuration
>   bool success = sim800l->setupGPRS(APN);
>   while(!success) {
>     success = sim800l->setupGPRS(APN);
>     delay(5000);
>   }
>   Serial.println(F("GPRS config OK"));
> }````
> ```

This  Library is amaizing thanks to the author. however, I am also stuck on the same issue, please kindly share the solution to this if you managed to sort this one out..!

ostaquet, 2022-01-04:
I'm not a fan of using String structure in embedded systems. The `sim800l->getDataReceived()`returns a char array (`char*`) and I have some doubts on the assignment to a String.

According to the [documentation on Arduino Reference](https://www.arduino.cc/reference/en/language/variables/data-types/stringobject/), the direct assignment can be done only from a const char array:

```
String stringOne = "Hello String";                    // using a constant String
String stringTwo = String("This is a string");        // converting a constant string into a String object
```

I would convert it through the constructor:
```
String input(sim800l->getDataReceived());
```

Not tested, to be validated.