ostaquet/Arduino-SIM800L-driver #51: Issue with custom url with GET request

atif-afridi, 2021-11-04:
hey , I am using the lib for sending data over http it was successful with your example ling i.e https://postman-echo.com/get?foo1=bar1&foo2=bar2. but I am unable to post to my custom provided url which is working on postman and I am able to open it on browser. 
since today morning the api used by your example isn't working for me. 

Error code I am receiving : 400 and 601 

please let me know anything further needed from my side.

I would really appreciate your help.

Thanks in advance.

ostaquet, 2021-11-07:
Error 400 means bad request.
Error 408 means request timeout (see also [https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/408|HTTP status 408])
Error 601 seems to be an error coming from the SimCom module.

When checking the documentation, you will find in [https://github.com/ostaquet/Arduino-SIM800L-driver/blob/master/extras/SIM800%20Series_AT%20Command%20Manual_V1.09.pdf|the datasheet] that the error 601 is related to a network error.

So, I guess that your URL and/or HTTP request is not valid and you have some connectivity issues.

atif-afridi, 2021-11-08:
thanks for your quick response.

yes may be we are having issues with that but the url response can be viewed on postman and in browser i.e chrome.

I have below points 

1 - https://postman-echo.com/get?foo1=bar1&foo2=bar2

above url was working fine but now its now working for me.

2- how can one debug the sim800L module is working correctly or not as I am seeing this command repeatedly

AT+CSQ

3- our custom url response can be viewed in postman and chrome browser but not by the module.

I am really hoping to get my work done with your lib as I have tried other solutions but nothing worked for me.
really appreciate your response.

thanks a lot.

ostaquet, 2021-12-11:
According to your screenshot, there is a weird behaviour by having some commands sent multiple times (namely ```AT+CREG?```).

Can you enable the full debug and post the result?

To do it, just update the constructor in the example:
```sim800l = new SIM800L((Stream *)serial, SIM800_RST_PIN, 200, 512);```

Has to be modified to:
```sim800l = new SIM800L((Stream *)serial, SIM800_RST_PIN, 200, 512, (Stream *)&Serial);```

closed 2021-12-29 by ostaquet