ostaquet/Arduino-SIM800L-driver #55: Library supported in ESP32 or not?

SurajitGithub1234, 2022-05-23:
Will this library supports ESP32?

SurajitGithub1234, 2022-05-23:
I tried to get the firmware version of the SIM800L module with the AT Command: AT+GMR and I got this response: Revision:1418B04SIM800L24. Can this module access HTTPS with this firmware variant?

ostaquet, 2023-05-05:
Hi @SurajitGithub1234,

As stated in https://github.com/ostaquet/Arduino-SIM800L-driver/issues/4, the library doesn't support ESP8266 due to the watchdog. I'm expecting this is the same problem with the ESP32.

Regarding the revision of your module, everything is explained. All revisions are supported by the lib. As stated in the README.md, you need the R14 to have the HTTPS but notice the security concerns about the SSL layer in the README.md.

KR

closed 2023-05-05 by ostaquet