ostaquet/Arduino-SIM800L-driver #60: issue with http get

Rothammel, 2023-01-17:
i am only able to send one HTTP get, on the second try I receive: Unable to init HTTP

```
SIM800L : Reset
SIM800L : Prepare internal buffer of 300 bytes
SIM800L : Prepare reception buffer of 512 bytes
SIM800L : Send "AT"
SIM800L : Receive timeout
Problem to initialize AT command, retry in 1 sec
SIM800L : Send "AT"
SIM800L : End of transmission
SIM800L : Receive "AT
OK
"
Setup Complete!
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ
+CSQ: 0,0
"
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ
+CSQ: 0,0
"
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ
+CSQ: 0,0
"
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ
+CSQ: 0,0
"
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ
+CSQ: 0,0
"
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ
+CSQ: 0,0
"
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ
+CSQ: 0,0
"
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ
+CSQ: 0,0
"
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ
+CSQ: 0,0
"
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ
+CSQ: 0,0
"
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ
+CSQ: 0,0
"
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ
+CSQ: 0,0
"
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ
+CSQ: 0,0
"
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ
+CSQ: 0,0
"
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ
+CSQ: 0,0
"
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ
+CSQ: 0,0
"
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ
+CSQ: 0,0
"
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ
+CSQ: 0,0
"
SIM800L : Send "AT+CSQ"
SIM800L : End of transmission
SIM800L : Receive "AT+CSQ
+CSQ: 17,0
"
Signal OK (strenght: 17)
SIM800L : Send "AT+CREG?"
SIM800L : End of transmission
SIM800L : Receive "AT+CREG?
+CREG: 0,2
"
SIM800L : Send "AT+CREG?"
SIM800L : End of transmission
SIM800L : Receive "AT+CREG?
+CREG: 0,2
"
SIM800L : Send "AT+CREG?"
SIM800L : End of transmission
SIM800L : Receive "AT+CREG?
+CREG: 0,2
"
SIM800L : Send "AT+CREG?"
SIM800L : End of transmission
SIM800L : Receive "AT+CREG?
+CREG: 0,5
"
Network registration OK
SIM800L : Send "AT+SAPBR=3,1,"Contype","GPRS""
SIM800L : End of transmission
SIM800L : Receive "AT+SAPBR=3,1,"Contype","GPRS"
OK
"
SIM800L : Send "AT+SAPBR=3,1,"APN","pinternet.interkom.de""
SIM800L : End of transmission
SIM800L : Receive "AT+SAPBR=3,1,"APN","pinternet.interkom.de"
OK
"
GPRS config OK
SIM800L : Send "AT+SAPBR=1,1"
SIM800L : End of transmission
SIM800L : Receive "AT+SAPBR=1,1
OK
"
GPRS connected with IP SIM800L : Send "AT+SAPBR=2,1"
SIM800L : End of transmission
SIM800L : Receive "AT+SAPBR=2,1
+SAPBR: 1,1,"10.36.99.191"
"
10.36.99.191
Start HTTP GET...
http://stan.selfhost.eu:5055/?id=123456&sat=11&hdop=0&lat=50.879913&lon=13.529003&altitude=473.7&speed=0.30
GPRS connected:  SIM800L : Send "AT+SAPBR=2,1"
SIM800L : End of transmission
SIM800L : Receive "AT+SAPBR=2,1
+SAPBR: 1,1,"10.36.99.191"
"
1
SIM800L : Send "AT+HTTPINIT"
SIM800L : End of transmission
SIM800L : Receive "AT+HTTPINIT
OK
"
SIM800L : Send "AT+HTTPPARA="CID",1"
SIM800L : End of transmission
SIM800L : Receive "AT+HTTPPARA="CID",1
OK
"
SIM800L : Send "AT+HTTPPARA="URL","http://stan.selfhost.eu:5055/?id=123456&sat=11&hdop=0&lat=50.879913&lon=13.529003&altitude=473.7&speed=0.30""
SIM800L : End of transmission
SIM800L : Receive "AT+HTTPPARA="URL","http://stan.selfhost.eu:5055/?id=123456&sat=11&hdop=0&lat=50.879913&lon=13.529003&altitude=473.7&speed=0.30"
OK
"
SIM800L : Send "AT+HTTPPARA="REDIR",1"
SIM800L : End of transmission
SIM800L : Receive "AT+HTTPPARA="REDIR",1
OK
"
SIM800L : Send "ATI"
SIM800L : End of transmission
SIM800L : Receive "ATI
SIM800 R14.18
"
SIM800L : initiateHTTP() - Support of SSL enabled
SIM800L : Send "AT+HTTPSSL=0"
SIM800L : End of transmission
SIM800L : Receive "AT+HTTPSSL=0
OK
"
SIM800L : Send "AT+HTTPACTION=0"
SIM800L : End of transmission
SIM800L : Receive "AT+HTTPACTION=0
OK
"
SIM800L : End of transmission
SIM800L : Receive "
+HTTPACTION: 0,200,0
"
SIM800L : readHTTP() - HTTP status 200
SIM800L : readHTTP() - Data size received of 0 bytes
SIM800L : Send "AT+HTTPREAD"
SIM800L : End of transmission
SIM800L : Receive "AT+HTTPREAD
OK
"
HTTP GET error 705
SIM800L : Send "AT+SAPBR=2,1"
SIM800L : End of transmission
SIM800L : Receive "AT+SAPBR=2,1
+SAPBR: 1,1,"10.36.99.191"
"
SIM800L : Send "AT+SAPBR=2,1"
SIM800L : End of transmission
SIM800L : Receive "AT+SAPBR=2,1
+SAPBR: 1,1,"10.36.99.191"
"
SIM800L : Send "AT+SAPBR=2,1"
SIM800L : End of transmission
SIM800L : Receive "AT+SAPBR=2,1
+SAPBR: 1,1,"10.36.99.191"
"
SIM800L : Send "AT+SAPBR=2,1"
SIM800L : End of transmission
SIM800L : Receive "AT+SAPBR=2,1
+SAPBR: 1,1,"10.36.99.191"
"
SIM800L : Send "AT+SAPBR=2,1"
SIM800L : End of transmission
SIM800L : Receive "AT+SAPBR=2,1
+SAPBR: 1,1,"10.36.99.191"
"
SIM800L : Send "AT+SAPBR=2,1"
SIM800L : End of transmission
SIM800L : Receive "AT+SAPBR=2,1
+SAPBR: 1,1,"10.36.99.191"
"

+SAPBR: 1,1,"10.36.99.191"
"
Start HTTP GET...
http://stan.selfhost.eu:5055/?id=123456&sat=11&hdop=0&lat=50.879929&lon=13.529037&altitude=472.2&speed=0.35&timestamp=1673944815
GPRS connected:  SIM800L : Send "AT+SAPBR=2,1"
SIM800L : End of transmission
SIM800L : Receive "AT+SAPBR=2,1
+SAPBR: 1,1,"10.36.99.191"
"
1
SIM800L : Send "AT+HTTPINIT"
SIM800L : End of transmission
SIM800L : Receive "AT+HTTPINIT
ERROR
"
SIM800L : initiateHTTP() - Unable to init HTTP
HTTP GET error 701
SIM800L : Send "AT+SAPBR=2,1"
SIM800L : End of transmission
SIM800L : Receive "AT+SAPBR=2,1
+SAPBR: 1,1,"10.36.99.191"
"
SIM800L : Send "AT+SAPBR=2,1"
SIM800L : End of transmission
SIM800L : Receive "AT+SAPBR=2,1
+SAPBR: 1,1,"10.36.99.191"
"
```

ostaquet, 2023-05-05:
@EndruK had a communication issue with the module and it solved it by forcing the baud rate for the AT communication.

I planned to integrate his suggestion in the library when I will have a module in hands.

Can you try this : https://github.com/ostaquet/Arduino-SIM800L-driver/issues/48#issuecomment-1064197833 ?

Rothammel, 2023-10-20:
Unfortunately this issue still exists:

`Problem to initialize AT command, retry in 1 sec
Setup Complete!
Signal OK (strenght: 17)
Network registration OK
GPRS config OK
GPRS connected with IP 10.65.225.171
Start HTTP GET...
HTTP GET error 408
HTTP GET error 701
HTTP GET error 701
GPRS disconnected !
Module in low power mode`

ostaquet, 2024-06-02:
As mentioned in the README, the project is no longer maintained as my module is broken. Thanks for reporting the issue.

closed 2024-06-02 by ostaquet