ostaquet/Arduino-SIM800L-driver #65: readHTTP() - Invalid answer on HTTP read

KolkaCool, 2024-04-28:
Greetings to the Warriors of Coding!
i encountered a problem when trying to use  HTTPS_GET_SoftwareSerial example code.

I am using Arduino Uno and SIM900 module, in the "HTTP GET" part, it returns ERROR 703, while in log data  i can see  +HTTPACTION:0,200,7438, wich as i understand that means 
that  the request has succeeded.

 I tried with example HTTP url and my server's  url. I can get data from my server manually - with AT commands, so i know it is working well with my modem.

Part of the log:

SIM800L : End of transmission
SIM800L : Receive "
+HTTPACTION:0,200,7438
"
SIM800L : readHTTP() - Invalid answer on HTTP read
703
HTTP GET error 703
SIM800L : Send "AT+SAPBR=0,1"
SIM800L : End of transmission
SIM800L : Receive "
OK
"

Thanks!

ostaquet, 2024-04-29:
Hey 👋 ,

Interesting! It seems that the SIM900 module is formatting the answer to AT command slightly differently.

As you can see in the code (`SIM800L.cpp` line 218) during the extract of the information:
```
  // Extract status information
  int16_t idxBase = strIndex(internalBuffer, "+HTTPACTION: ");
  if(idxBase < 0) {
    if(enableDebug) debugStream->println(F("SIM800L : readHTTP() - Invalid answer on HTTP read"));
    return 703;
  }
```

The extract is based on the answer starting with `+HTTPACTION: ` (ending with a blank space). According to your logs, the `+HTTPACTION:` doesn't end with blank space.

A slightly adjustment of the code should work in your case by ignoring the blank space on line 219.

Let me know if you figured out.

ostaquet, 2024-06-02:
As mentioned in the README, the project is no longer maintained as my module is broken. Thanks for reporting the issue.

closed 2024-06-02 by ostaquet